Bodmer/TFT_eSPI #2653: the esp32 crashes when the wifi connection is lost

woodymgh, 2023-05-12:
1. A description of the problem and the conditions that cause it to occur

if the wfi connection is lost while the display is being drawn on, the esp32 will crash. the higher the writing load on the display, the greater the likelihood of a crash. To reproduce this I created a simple program that creates a sprite and moves it across the display.
Not every disconnection leads to a system crash.

2. IDE (e.g. Arduino or PlatformIO)

I use PlatformIO

3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
2.4.78
2.5.30

4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)

5. Procesor, e.g RP2040, ESP32 S3 etc
ESP32 WROOM

6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
ST7789

7. Interface type (SPI or parallel)
SPI (HSPI)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // ST7789 240 x 240
#define TFT_MISO -1
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   2  // Chip select control pin
#define TFT_DC    12  // Data Command control pin
#define TFT_RST   27  // Reset pin (could connect to RST pin)

2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
see 3.

3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.

[Archiv.zip](https://github.com/Bodmer/TFT_eSPI/files/11468221/Archiv.zip)

4. Screen shot pictures showing the problem  (just drag and drop in message window)
----

tft-code main.c
`
TFT_eSPI tft = TFT_eSPI();  // Invoke custom library
TFT_eSprite ball=TFT_eSprite(&tft);
`

The Sprite:
`
	ball.createSprite(21,21);
	ball.fillRect(0,0,20,20,TFT_BLACK);
	ball.fillCircle(10,10,7,TFT_GREEN);
`

The loop (timed):

`
tft.setViewport(0,150,110,50);
ball.pushSprite(xball,30);
tft.resetViewport();
`
For details see the Example in the zip-file.
The example uses:
wifimanager for init of the Wifi
Async Webserver
Arduino OTA
telnetspy (disabled)

Bodmer, 2023-05-14:
I will see if I can adapt your code for the Arduino environment and run it. I suspect this is not a TFT_eSPI issue. Have you decoded and examined the crash report? Have you checked the Espressif board package repository issue list?

Bodmer, 2023-05-14:
It looks like this is related to your issue:
https://stackoverflow.com/questions/48851110/esp32-stable-wifi-connection-on-reboot-router-esp32-lost-connection-bads-conne

woodymgh, 2023-05-14:
Unfortunately there is no crash report. The esp crashes so that neither the watchdog is active nor a crash message appears. in the complete project I have already removed everything that could be the trigger. what is left is the display. After a lot of research I couldn't find a solution. So it's not about getting a reconnect, there are some solutions for this, but about the esp32 crashing. my fear is/was that using the spi or dma with the connection lost would lead to a crash due to the dual processor. locking may be necessary here, but I don't know where (if it's the problem at all).

Bodmer, 2023-05-14:
Increase the core debug level until meaningful crash output is reported. See Espressif documentation.

Bodmer, 2023-05-21:
I had a look but I get too many compile errors. I guess I have different libraries loaded.

It looks like you are using this library which has a lot of issues with not action or replies, so maybe that is a good place to investigate:
https://github.com/me-no-dev/ESPAsyncWebServer

I am not able to help as I suspect this is not a TFT_eSPI issue.

closed 2023-05-21 by Bodmer

woodymgh, 2023-06-17:
First of all thank you very much for your support. Unfortunately, it took a while to get my suspicions about the error confirmed. Long story short, it has nothing to do with the library, but does have to do with the LCD display.

The ESP is connected to a DC/DC converter here. Whenever the connection to the WLAN is lost, the power consumption of the ESP is extremely high. If there are additional activities on the display, the power requirement rises above what the DC/DC converter can provide. The consequences are unforeseeable crashes of the ESP without log output. I adjusted the DC/DC converter. Now everything works as desired.