Bodmer/TFT_eSPI #2662: ST7796S issue

valkowe, 2023-05-19:
Hello Bodmer, 

Thank you for creating this library as it's done wonders for me! 

Unfortunately recently I have been having issues with ST7796S, does the library support it? Getting like 30% of the screen working and the other part is white/no signal. I can attach pictures if the ST7796S is supported and if I need to do adjustments to my setup.

Thanks in advance! 

Using IDE
2.5 version of lib
ESP32

Bodmer, 2023-05-19:
I will connect up one of those display types and see if it works.

Post a link to the display vendors web page.

Bodmer, 2023-05-21:
I have connected up an ST7796 screen and it is working fine.

Run the Read_User_Setup sketch, check the serial output and and post it here.

Example:
```
TFT_eSPI ver = 2.5.30
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7796
Display width  = 320
Display height = 480

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 21

TFT_BL           = GPIO 32
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50

```

valkowe, 2023-05-23:
Thank you so much for your reply! I believe it was a hardware issue as I somehow managed to fix it with new display and ESP32, still not sure what was the issue - either ESP of display was faulty I think, as I bought new ones and they work. Once again thank you for your reply and for the amazing library you have created!

closed 2023-05-23 by valkowe