Bodmer/TFT_eSPI #2663: Crash when printing text/reading font data from flash?

doctea, 2023-05-20:
>1. A description of the problem and the conditions that cause it to occur

Crashing when printing strings on XIAO RP2040 in a project that was previously stable and working; possibly due to upgrading toolchain or platform packages; at other times, strange things like lines drawing correctly but text not rendering at all (nothing output to screen).  Reverting project and library code is still producing binaries that fail.

From debugging and single-stepping through the code, crash is happening during pgm_read_word() trying to read from a gfxFont pointer to 0x69617661.

>2. IDE (e.g. Arduino or PlatformIO)

vscode + platformio.

> 3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu

Tried most recent commit version, also tried reverting to previous versions from February, with apparently same results.

> 4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu

PACKAGES:
 - framework-arduinopico @ 1.30101.0+sha.e95248a
 - tool-rp2040tools @ 1.0.2
 - toolchain-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)

>5. Procesor, e.g RP2040, ESP32 S3 etc

Seeed XIAO RP2040.

> 6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.

ST7789, but I don't think it actually gets that far (rendering to a sprite first), and it was also working perfectly for several months until I ran into problems.

> 7. Interface type (SPI or parallel)

SPI

> Plus further information as appropriate to the problem:
> 2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)

Relevant build flags from platformio.ini:

```
build_flags = 
	-DRP2040_PIO_SPI
	-DDISABLE_ALL_LIBRARY_WARNINGS
	-DSCREEN_ROTATION=3
	-DTFT_SCLK=D8
	-DTFT_MOSI=D10
	-DTFT_MISO=-1
	-DTFT_CS=D7
	-DTFT_RST=-1
	-DTFT_DC=D9
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DTFT_WIDTH=135
	-DTFT_HEIGHT=240
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_FONT8N=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=50000000
```

> 3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.

Current project is huge but complicated, but I'll attempt to put together a smaller demonstration of the problem, if necessary and if I succeed in making a simpler test version that can replicate the problem!  Am hoping that the odd location in memory of the gfxfont will be a big enough clue!

Thanks for any insight you can offer, I've been all over the place trying to pin down where responsibility for this problem lies without much success so far!

doctea, 2023-05-20:
Hmm, well, this seems strange and a little embarrassing, but adding a call to setFreeFont() to my initialisation seems to have solved this problem.  But I don't remember this ever being a part of my code before, and things were working intermittently before I added this and I can't find any mention of it in my previous repository commits..!

Bodmer, 2023-05-21:
Glad to hear it is solved!

closed 2023-05-21 by Bodmer