Bodmer/TFT_eSPI #2664: Warnings when adding buttons

fungreenfox, 2023-05-21:
Esp32 Wroom

TFT_Espi library version 2.3.7
Someone said a lot about having more trouble with newer versions of this library, so it isnt updated.
Has this warning been fixed in a later version?

warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ESCROn_btn.initButton(&tft, 35, 155, 55, 35, BLACK, GREY, GREEN, "Y", 3);
ESCROn_btn.initButton(&tft, 35, 155, 55, 35, BLACK, GREY, GREEN, "Y", 3);
    ESCROff_btn.initButton(&tft, 100, 155, 55, 35, BLACK, GREY, RED, "N", 3);
    readESCR();
    if (ESCR == 0)
    {
      tft.setTextSize(2); tft.setTextFont(1);
      tft.setTextColor(DKGREY, MenuBGColor);
      tft.setCursor(150, 145);
      tft.print(F("Show Station Frequency"));
      ESCROn_btn.drawButton(false);
      ESCROff_btn.drawButton(true);
    }

fungreenfox, 2023-05-21:
Found this one: https://github.com/Bodmer/TFT_eSPI/issues/1745
It does not tell me, if it was fixed as a 1745 issue.

Bodmer, 2023-05-22:
You can ignore the warnings. An update is planned.

fungreenfox, 2025-07-20:
No worry, i have corrected this in two of the librarys, inserting const before an expression. Warnings gone, but if getting an update, it might be overwritten again.