Bodmer/TFT_eSPI #2666: Screen goes white, when radio initialised

fungreenfox, 2023-05-22:
TFT screen with TFT_Espi library.
When the radio is initialised from setup, the TFT screen ili9488 goes white. Touch is working, but any color goes white.
This all worked ok with an older radio.h library.

#define        T_DO    19
#define        T_DIN   23
#define        T_CS    5                
#define        T_CLK   18 
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15 //for st3577 use pin 5 Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_BL   25  // LED back-light (required for M5Stack)
#define TOUCH_CS 5   // Chip select pin (T_CS) of touch screen

Bodmer, 2023-05-22:
Assuming the TFT_eSPI library examples work OK then this is likely to be a resource conflict with the "radio" library. This is something you will need to debug yourself, or continue to use the older version.  Maybe the radio library author can help.

closed 2023-05-22 by Bodmer

fungreenfox, 2023-06-09:
For the display, how many different pins are possible to have (one at each time), like -1 and D4 (esp32), and where is -1 possitioned?