Bodmer/TFT_eSPI #2670: Touch not working using esp32-2432s028 (ILI9341, ESP32-WROOM-32, XPT2046)

Atmannen, 2023-05-26:
I am trying to get the wifi marauder to work with this board: esp32-2432s028. (https://www.makerfabs.com/sunton-esp32-2-8-inch-tft-with-touch.html) and everything works except touch. After lots of debugging I tried using the Test_Touch_Controller sketch using Arduino IDE and I only see zeroes as x,y,z. No response at all when using touch. I have tried the LovyanGFX library and that works.

So after some heavy searching I found a fork (https://github.com/Seeed-Studio/Seeed_Arduino_LCD) of this repo where they have added  support to specify MISO, MOSI etc for the touch. I have trying to use that fork instead but I get lots of errors (I will continue to work on that).

Do anyone understand why this repository does not register touch using this board while other libraries do? Can the reason be that the MISO, MOSI etc is not specified here? 

User_Setup.h
`
#define ILI9341_DRIVER
#define TFT_MISO 12 // Matching T_DO
#define TFT_MOSI 13 // Matching T_DIN
#define TFT_SCLK 14 // Matching T_CLK
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   2  // Data Command control pin
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TFT_BL   21  // LED back-light (only for ST7789 with backlight control pin)
#define TOUCH_CS 33     // Chip select pin (T_CS) of touch screen
#define SPI_FREQUENCY  40000000 // Maximum to use SPIFFS
#define SPI_READ_FREQUENCY  16000000
// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  1000000

#define USE_HSPI_PORT
`

StrandedAlien, 2023-05-26:
Maybe you want to try using a pins.h declaration file, like so:

https://github.com/Xinyuan-LilyGO/T-HMI/blob/master/examples/touch_calibration/pins.h

Bodmer, 2023-05-28:
The TFT_eSPI library expects the touch controller and TFT to be on the same SPI bus.  The sunton board connects the touch controller to diferent pins:
```
#define LCD_MOSI 13
#define LCD_MISO 12
#define LCD_SCK 14
#define LCD_CS 15
#define LCD_RST -1
#define LCD_DC 2
#define LCD_BL 21

#define TOUCH_MOSI 32
#define TOUCH_MISO 39
#define TOUCH_SCK 25
#define TOUCH_CS 33
#define TOUCH_IRQ 36
```
The TFT_eSPI library can still be used but you will need an independant library for the touch functions.  There are quite a few libraries that support the XPT2046 touch controller, such as this one:
https://github.com/achillhasler/TFT_eTouch

There are no plans to change the TFT_eSPI library to support different touch controllers or different I/O pins. In the longer term the plan is to deprecate/remove touch functionality and create seperate librarys so that other touch chips can be supported easily (capacitance FTxxxx controlers etc.)