Bodmer/TFT_eSPI #2674: LCD with GC9A01 ESP32 suddenly not working

mwlodarr, 2023-05-28:
Hello,
I was using this libbrary in project based on https://github.com/VolosR/BoatGauges, it was working fine until 3 days ago. I have changed User_setup_select to use with RP2040, on the next day I changed it back to use with ESP32, after that LCD is not displaying images. Only tft.fillScreen with solid color work, program is compiling fine without any issues, but nothing except changing tft_fillScreen work. I have tried compiling program on my previous notebook which I was using to programming, and it works fine, everything on LCD is working like a charm. Code in main loop is working because it is printing in serial, but it seems like no react to img.pushImage and img.pushSprite. I have tried reinstalling Arduino IDE, libraries, older libraries version, older Arduino IDE, also I have tried copying library from old notebook where it works to my new notebook. I have also tried install everything on another PC and it is also not working. I have run Read_user_setup, on both notebooks it is printing same values. I have no idea what I can do next, wiring is ok because with old notebook it's working, code is also ok because I am using exactly the same code on both notebooks. Arduino IDE 2.1.0 and 1.8.13 I do not attach code because now when I try to fix this, I am using default boatgauges code with only changed backlight pin, button pin and analog input pin.

output from diagnostic on new notebook
'
[code]
TFT_eSPI ver = 2.4.79
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9A01
Display width  = 240
Display height = 320

MOSI    = GPIO 15
SCK     = GPIO 4
TFT_CS   = GPIO 16
TFT_DC   = GPIO 17
TFT_RST  = GPIO 5

TFT_BL           = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]
'

output from old notebook 
'
[code]
TFT_eSPI ver = 2.4.79
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9A01
Display width  = 240
Display height = 320

MOSI    = GPIO 15
SCK     = GPIO 4
TFT_CS   = GPIO 16
TFT_DC   = GPIO 17
TFT_RST  = GPIO 5

TFT_BL           = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]
'

Bodmer, 2023-05-28:
It appears you have done a lot to get it working again. It is always tricky to help with these "it was working but has stopped working" issues and so any help will be guesswork and suggestions.

Have you checked for differences in the ESP32 board package version installed?
Are there any differences in the IDE menu option settings (board selected etc)?
If the images used are in SPIFFS or LittleFS then different ESP32 board packages sometimes require complete upload of both images AND sketch to work correctly.

Since it works with one laptop with the same ketch and setup I do not think this is an issue with the TFT_eSPI.

mwlodarr, 2023-05-28:
Thanks for help, I Have checked version in board manager and old laptop was 1.0.6 version of esp32 library, downgroaded on new laptop and it helped, now it is working. Thank You very much.

Bodmer, 2023-05-28:
Great, glad you have it working. Later ESP32 board packages introduced some incompatibilities that required updates to the TFT_eSPI library.

closed 2023-05-28 by Bodmer