Bodmer/TFT_eSPI #2682: Display P023T009-V2 in 8-bit interface mode with driver ILI9342C - black screen

brightproject, 2023-06-02:
I use display P023T009-V2 
[image]
with stm32f401ccu6 with library [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)
Connection according to the datasheet on the display and connection as in MCUFRIEND_kbv
[image]
```
    //LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
    //STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**
```
The display is switched to 8-bit mode by connecting pin 24(w16_8x) to ground(IM1 -> LOW).
[image]

TFT_graphicstest_one_lib.ino
```
TFT_eSPI library test!
Benchmark                Time (microseconds)
Screen fill              963641
Text                     42001
Lines                    554744
Horiz/Vert Lines         79712
Rectangles (outline)     46658
Rectangles (filled)      2327901
Circles (filled)         355437
Circles (outline)        203002
Triangles (outline)      119031
Triangles (filled)       841875
Rounded rects (outline)  97113
Rounded rects (filled)   2377060
Done!
```
Test by MCUFRIEND_kbv LCD_ID_readnew.ino
```
Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00.ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04.Manufacturer ID
reg(0x0009) 00 00 00 00 00.Status Register
reg(0x000A) 00 08.Get Power Mode
reg(0x000C) 04 04.Get Pixel Format
reg(0x0030) 00 00 00 00 00.PTLAR
reg(0x0033) 00 00 00 00 00 00 00.VSCRLDEF
reg(0x0061) 00 00.RDID1 HX8347-G
reg(0x0062) 00 00.RDID2 HX8347-G
reg(0x0063) 00 00.RDID3 HX8347-G
reg(0x0064) 04 04.RDID1 HX8347-A
reg(0x0065) 04 04.RDID2 HX8347-A
reg(0x0066) 04 04.RDID3 HX8347-A
reg(0x0067) 04 04.RDID Himax HX8347-A
reg(0x0070) 00 00.Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00.RD_DDB SSD1963
reg(0x00B0) 00 40.RGB Interface Signal Control
reg(0x00B3) 00 00 18 18 18.Frame Memory
reg(0x00B4) 04 44.Frame Mode
reg(0x00B6) 04 0E 84 1D 04.Display Control
reg(0x00B7) 04 07.Entry Mode Set
reg(0x00BF) 04 06 06 06 06 06.ILI9481, HX8357-B
reg(0x00C0) 00 B2 B2 B2 B2 B2 B2 B2 B2.Panel Control
reg(0x00C1) 00 00 00 00.Display Timing
reg(0x00C5) 04 04.Frame Rate
reg(0x00C8) 00 50 02 02 02 02 02 06 06 06 06 06 06.GAMMA
reg(0x00CC) 04 04.Panel Control
reg(0x00D0) 00 00 00 00.Power Control
reg(0x00D1) 00 00 00 00.VCOM Control
reg(0x00D2) 00 11 00.Power Normal
reg(0x00D3) 00 00 93 42.ILI9341, ILI9488
[... 188 lines cut ...]

// Write the display data into GRAM here
// LCD_ILI9342_CMD(0x2C);//GRAM start writing
for ( i=0; i<240; i++)
for ( j=0; j<320; j++)
LCDDATA_Write(0xdata); //write display data

void LCD_Enter Sleep_ILI9342(void)
{
  LCD_ILI9342_CMD(0x28);//Display off
  LCD_ILI9342_CMD(0x10);//Enter Sleep mode
}

void LCD_Exit Sleep_ILI9342(void)
{
  LCD_ILI9342_CMD(0x11);//Exit Sleep
  Delayms(120);
  LCD_ILI9342_CMD(0x29);//Display on
}
```

ILI9342C:
[ILI9342C_DTS_V102_20120328(1).pdf](https://github.com/Bodmer/TFT_eSPI/files/11633842/ILI9342C_DTS_V102_20120328.1.pdf)
P023T009-V2 SPEC:
[P023T009-V2 SPEC(2).pdf](https://github.com/Bodmer/TFT_eSPI/files/11633848/P023T009-V2.SPEC.2.pdf)

Bodmer, 2023-06-02:
Try a different init sequence, see here:
https://github.com/Bodmer/TFT_eSPI/issues/1172

brightproject, 2023-06-03:
> Try a different init sequence, see here:

I've tried changing the parameters of the power commands, but didn't get any results.
I took the code sent by the Chinese and found the necessary lines in the ILI9341_Init.h file using the commands and replaced some - this also did not work.
I decided to compare the datasheets for ILI9341 and ILI9342C
I was surprised that ILI9342 does not have a physical contact EXTC
[image]
And how to initialize the display, I do not understand yet.
And therefore, the display is apparently not initialized.
I tried adding code
```
  writecommand(0xC8);
  writedata(0xFF);
  writedata(0x92);
  writedata(0x42);
```
to ILI9341_Init.h , but again I got nothing but a white screen.
Connected all contacts, this true, because the parameters are read correctly, therefore CS DC WR RD RST control signals are correctly transmitted and data from the ILI9342C controller is transmitted and received via D0-D7.
Why isn't there an image, no image at all, not blurry, not silhouettes, not gamma, not stripes?
In the datasheet ILI9342 I found that the 8080 interface is I and II.
[image]
I had suspicions that the Chinese made a display and made a mistake with the designations D0-D7 and D9-D15 in the datasheet.
But after all, there is a communication with the display controller via D0-D7 and I receive data.
The situation is not very clear to me.
Plus to these issues, I can't use roger clarke's kernel to check code for *

brightproject, 2023-06-03:
I create same problem discussion in fork stm32core
https://github.com/stm32duino/Arduino_Core_STM32/discussions/2035

Bodmer, 2023-06-05:
Have you got the Mcufriend_kbv library running with that hardware?

brightproject, 2023-06-05:
> Have you got the Mcufriend_kbv library running with that hardware?
Mcufriend_kbv library doesn't compile for stm32f401...
I I think the problem is with the initialization of the display on the driver ILI9342C
EXTC and soft reset - you wrote about the reset as the fact that on some displays the hard reset signal does not reach.
But I have tried the commands:
```
    writecommand(0x01); // SW reset
    delay(120);

    writecommand(0x11); // Sleep out, also SW reset
    delay(120);
```
But nothing has changed.

Bodmer, 2023-08-02:
Usually:

#define ILI9342_DRIVER

works fine.

Bodmer, 2023-09-06:
Assumed resolved.

closed 2023-09-06 by Bodmer