Bodmer/TFT_eSPI #2687: RP2040 andTouchscreen always gives high Z values

ryanaukes, 2023-06-03:
Hello,

I am trying to use this library with my RP2040 microcontroller and ILI9341 screen with touch screen and SD reader.
Currently I only connected the TFT screen and Touschreen pins.

Connected as follows:
TFT VCC -> 5V of PSU
TFT VCC -> GND of PSU and RP2040
TFT CS -> pin 20
TFT RESET -> pin 19
TFT DC -> pin 18
TFT MOSI -> pin 3
TFT SCK -> pin 2
TFT LED -> 5V of PSU
TFT MISO -> pin 0
TFT T_CLK -> pin 2
TFT T_CS -> pin 21
TFT T_DIN -> pin 3
TFT T_DO -> pin 0

This is what the file called 'User_Setup_Select.h' looks like:

`// This header file contains a list of user setup files and defines which one the
// compiler uses when the IDE performs a Verify/Compile or Upload.
//
// Users can create configurations for different boards and TFT displays.
// This makes selecting between hardware setups easy by "uncommenting" one line.

// The advantage of this hardware configuration method is that the examples provided
// with the library should work with immediately without any other changes being
// needed. It also improves the portability of users sketches to other hardware
// configurations and compatible libraries.
//
// Create a shortcut to this file on your desktop to permit quick access for editing.
// Re-compile and upload after making and saving any changes to this file.

// Customised User_Setup files are stored in the "User_Setups" folder.

// It is also possible for the user tft settings to be included with the sketch, see
// the "Sketch_with_tft_setup" generic example. This may be more convenient for
// multiple projects.

#ifndef USER_SETUP_LOADED //  Lets PlatformIO users define settings in
                          //  platformio.ini, see notes in "Tools" folder.

///////////////////////////////////////////////////////
//   User configuration selection lines are below    //
///////////////////////////////////////////////////////

// Only ONE line below should be uncommented to define your setup.  Add extra lines and files as needed.

//#include <User_Setup.h>           // Default setup is root library folder

//#include <User_Setups/Setup1_ILI9341.h>  // Setup file for ESP8266 configured for my ILI9341
//#include <User_Setups/Setup2_ST7735.h>   // Setup file for ESP8266 configured for my ST7735
//#include <User_Setups/Setup3_ILI9163.h>  // Setup file for ESP8266 configured for my ILI9163
//#include <User_Setups/Setup4_S6D02A1.h>  // Setup file for ESP8266 configured for my S6D02A1
//#include <User_Setups/Setup5_RPi_ILI9486.h>        // Setup file for ESP8266 configured for my stock RPi TFT
//#include <User_Setups/Setup6_RPi_Wr_ILI9486.h>     // Setup file for ESP8266 configured for my modified RPi TFT
//#include <User_Setups/Setup7_ST7735_128x128.h>     // Setup file for ESP8266 configured for my ST7735 128x128 display
//#include <User_Setups/Setup8_ILI9163_128x128.h>    // Setup file for ESP8266 configured for my ILI9163 128x128 display
//#include <User_Setups/Setup9_ST7735_Overlap.h>     // Setup file for ESP8266 configured for my ST7735
//#include <User_Setups/Setup10_RPi_touch_ILI9486.h> // Setup file for ESP8266 configured for ESP8266 and RPi TFT with touch

//#include <User_Setups/Setup11_RPi_touch_ILI9486.h> // Setup file configured for ESP32 and RPi TFT with touch
//#include <User_Setups/Setup12_M5Stack_Basic_Core.h>// Setup file for the ESP32 based M5Stack (Basic Core only)
//#include <User_Setups/Setup13_ILI9481_Parallel.h>  // Setup file for the ESP32 with parallel bus TFT
//#include <User_Setups/Setup14_ILI9341_Parallel.h>  // Setup file for the ESP32 with parallel bus TFT
//#include <User_Setups/Setup15_HX8357D.h>           // Setup file for ESP8266 configured for HX8357D
//#include <User_Setups/Setup16_ILI9488_Parallel.h>  // Setup file for the ESP32 with parallel bus TFT
[... 413 lines cut ...]
// With a ST7735 display more than 27MHz may not work (spurious pixels and lines)
// With an ILI9163 display 27 MHz works OK.

// #define SPI_FREQUENCY   1000000
// #define SPI_FREQUENCY   5000000
// #define SPI_FREQUENCY  10000000
// #define SPI_FREQUENCY  20000000
// #define SPI_FREQUENCY  32000000
 #define SPI_FREQUENCY  20000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000`

When I upload the example sketch called 'TFT_Rainbow_one_lib', the screen works fine.

When I upload the example skethc called 'Touch_calibrate', the text and arrows pointing to the corners are displayed shortly, but it quickly continues to the part where it shows the text 'Touch screen to test!'. It seems like the screen is pressed, without me doing anything. 

When I run the example sketch called 'Test_Touch_Controller', I can see that the Z values are always really high (fluctuating between 500 and 7000, but mostly above 6000). 

When I run the same test sketches on a ESP32, it everything works fine, so the screen seems to be working. 
I hope someone can help me find the problem. 
Thanks in advance!

Bodmer, 2023-06-04:
This is not a behaviour I have seen before. Can you try reducing the touch SPI frequency to 1MHz.
```
#define SPI_TOUCH_FREQUENCY  1000000
```

Bodmer, 2023-06-04:
Another option is to try this library:
https://github.com/achillhasler/TFT_eTouch

copyCATandpasteDOG, 2023-06-04:
Hi ,yanaukes,
have you tried with MISO connected to nothing (on the air),even if the setup shows pin0?

ryanaukes, 2023-06-04:
> This is not a behaviour I have seen before. Can you try reducing the touch SPI frequency to 1MHz.
> 
> ```
> #define SPI_TOUCH_FREQUENCY  1000000
> ```

@Bodmer 
Thanks for the answer! Unfortunately that does not solve my proble. Is there anything else I can do to debug this problem?
Just to be sure: the pins are referring to the GP pins (in green), so not the numbers in the gray squares, according to [this](https://cdn-learn.adafruit.com/assets/assets/000/099/339/large1024/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004) picture?

> Hi ,yanaukes, have you tried with MISO connected to nothing (on the air),even if the setup shows pin0?

@copyCATandpasteDOG 
Thanks for the suggestion. I tried disconnecting the MISO pin, but then I get no response at all from the touchscreen.

ryanaukes, 2023-06-04:
Maybe it is helpfull. This is a part of the serial output of the "Test_Touch_Controller" sketch:

```
x: 0     y: 7936     z: 4863
x: 0     y: 7936     z: 4607
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 65535
x: 0     y: 7936     z: 767
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 2559
x: 2560     y: 7936     z: 64255
x: 0     y: 7936     z: 3583
x: 0     y: 7936     z: 65023
x: 0     y: 7936     z: 62719
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 1535
x: 0     y: 7936     z: 511
x: 0     y: 7936     z: 62719
x: 0     y: 7936     z: 62207
x: 0     y: 7936     z: 62207
x: 0     y: 7936     z: 62207
x: 0     y: 7936     z: 1279
x: 0     y: 7936     z: 5119
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 62719
x: 0     y: 7936     z: 3071
x: 0     y: 7936     z: 64767
x: 0     y: 7936     z: 3327
x: 0     y: 7936     z: 3071
x: 0     y: 7936     z: 62207
x: 0     y: 7936     z: 64767
x: 0     y: 7936     z: 4607
x: 0     y: 7936     z: 64255
x: 0     y: 7936     z: 511
x: 4864     y: 7936     z: 62463
x: 0     y: 7936     z: 1023
x: 0     y: 7936     z: 1023
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 1791
x: 0     y: 7936     z: 2815
x: 0     y: 7936     z: 767
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 65535
x: 0     y: 7936     z: 62463
x: 0     y: 7936     z: 4607
x: 0     y: 7936     z: 3071
x: 0     y: 7936     z: 62719
x: 0     y: 7936     z: 64255
x: 0     y: 7936     z: 5631
x: 0     y: 7936     z: 62207
x: 7168     y: 7936     z: 62207

```

copyCATandpasteDOG, 2023-06-05:
ryanaukes wrote:
"Thanks for the suggestion. I tried disconnecting the MISO pin, but then I get no response at all from the touchscreen."

of course  TFT T_DO -> pin 0, even if MISO is disconnected

Bodmer, 2023-06-05:
Which version of TFT_eSPI is being used?

Have you tried the suggested TFT_eTouch library?

copyCATandpasteDOG, 2023-06-06:
Maybe the pico has specifics pins that don't work in some cases,
This setup works well for me:

```
// Stripped down User_Setup.h version for the Pico-Matrix-Touch-Keyboard //

#define ILI9341_DRIVER     

#define TFT_MISO  0
#define TFT_MOSI  3
#define TFT_SCLK  2
#define TFT_CS   20  // Chip select control pin
#define TFT_DC   22  // Data Command control pin
#define TFT_RST  21  // Reset pin (could connect to Arduino RESET pin)

#define TOUCH_CS 14     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

```

The sketch that works for me with pico: "Big keypad" from TFT e_spi (keypad) :

```
//Big keypad

#include "Arduino.h"
#include "LittleFS.h"

#include <SPI.h>
#include <TFT_eSPI.h>

// Name of the Touch calibration file
#define CALIBRATION_FILE "/TouchCalData2"

// Define our filesystem
#define FILESYSTEM LittleFS

// Set to tru if you want the calibration to run on each boot
#define REPEAT_CAL false

// Keypad start position, key sizes and spacing
#define KEY_W 95 // Key width
#define KEY_H 95 // Key height
#define KEY_SPACING_X 20 // X gap
#define KEY_SPACING_Y 10 // Y gap

#define KEY_X (KEY_W/2) + KEY_SPACING_X // X-axis centre of the first key
#define KEY_Y (KEY_H/2) + KEY_SPACING_Y // Y-axis centre of the first key

#define KEY_TEXTSIZE 3   // Font size multiplier
[... 193 lines cut ...]
    tft.setTextFont(1);
    tft.println();

    if (REPEAT_CAL) {
      tft.setTextColor(TFT_RED, TFT_BLACK);
      tft.println("Set REPEAT_CAL to false to stop this running again!");
    }

    tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.println("Calibration complete!");

    // store data
    File f = FILESYSTEM.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calData, 14);
      f.close();
    }
  }
}
  
```

Hope you'll find something to do with

ryanaukes, 2023-06-06:
> Which version of TFT_eSPI is being used?
> 
> Have you tried the suggested TFT_eTouch library?

I am using version 2.5.0.
Unfortunately I have not had the time to test the TFT_eTouch library. Hopefully this weekend I have some more time to test it.

ryanaukes, 2023-06-06:
Today I had some time to do some more testing.
It seems that the problem is related to the  Arduino-mbed RP2040 core I am using.
I installed the  arduino-pico core by Earle Philhower using the board manager and now everything seems to work!

I have run into another problem though. I can’t make the JPEG SD card example work fort he pi pico. 
Should I open a new issue for this, since it is unrelated, or can I ask for help with that here?

Thanks again for your help!

copyCATandpasteDOG, 2023-06-07:
Good news,
i always thought you was using the Philhower library :)

Bodmer, 2023-12-07:
I have tested this and it is working now with the latest Github master (not released yet).

I have not made any changes specific to this issue but maybe some other updates fixed this.

closed 2023-12-07 by Bodmer