Bodmer/TFT_eSPI #2688: Compile error swap_coord' was not declared

videobuff, 2023-06-03:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
**I can no longer compile the attached sketch, which has been working for more than a your now.**
**I get the error that swap_coord' was not declared in this scope** 
3. IDE (e.g. Arduino or PlatformIO)
**ARDUINO on Mac Montery Version 2.1.0**

4. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
**2.5.0**
6. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
**2.0.9**
8. Procesor, e.g RP2040, ESP32 S3 etc
**ESP32**
10. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
**Setup25_TTGO_T_Display.h**
12. Interface type (SPI or parallel)
**SPI**
Plus further information as appropriate to the problem:
1. TFT to processor connections used
**I am using the LILYGO® TTGO T1 ESP-32 V1.3 Rev1**

4. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
**See below**
5. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
**See below**

Output error after compiling

```
/Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI 3.cpp: In member function 'void TFT_eSPI::fillTriangle(int32_t, int32_t, int32_t, int32_t, int32_t, int32_t, uint32_t)':
/Users/erikschott/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI 3.cpp:2513:5: error: 'swap_coord' was not declared in this scope
     swap_coord(y0, y1); swap_coord(x0, x1);

```

Dear Bodmer,
I have been using this sketch for a long time but needed to adapt it. Now, it wont compile anymore and stops at the swap_coord error. All libraries were updated.
The sketch is attached. You have to adapt the init.h file for your local situation. Without MQTT server you still get a picture.

[Webrotor_Antenna_Sensor_TTGO.zip](https://github.com/Bodmer/TFT_eSPI/files/11643413/Webrotor_Antenna_Sensor_TTGO.zip)

[TFT_SPI setup.zip](https://github.com/Bodmer/TFT_eSPI/files/11643426/TFT_SPI.setup.zip)

Bodmer, 2023-06-03:
Looks like parts from different versions of the library have been patched together in your environment.  Where does file 3.cpp come from?

The swap_coord()  function does not exist in 2.5.0, it has been replaced with the more generically named funtion transpose():
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L981-L983

closed 2023-06-03 by Bodmer

videobuff, 2023-06-04:
So stupid from me. Too long ago i worked on it. Removed yiour library, reinstalled it, did the settings en working. Thanks you.