Bodmer/TFT_eSPI #2697: Radio with display ili9488

fungreenfox, 2023-06-09:
Building a stationary radio with esp32 wroom 30 pin, and display ili 9488 3.5". 
Problem is, that when the Si4703 radio module is initialized seconds later after the tft.ini(), the display goes all white.
It appears to react upon the touch, so it is somehow active. The reversing doesnt affect.
There is a spiffs calibration running if wanted, and it doesnt go white background when calibratrion is going on.
The latest tft_espi libs are not used, since someone said there was trouble with it, and this older version appears to work with lesser implementations.
How to solve the display issue?
Script is complicated, and attached as links.
dropbox arduino esp32 radio 2c
https://www.dropbox.com/s/5jcuh6wvon71o4a/arduino_radio7_Si4703_V1_Tiny_RTC_copy2c.zip?dl=0
https://www.dropbox.com/s/arvvz5odn1p30hp/TFT_eSPI.zip?dl=0

Bodmer, 2023-07-15:
It is difficult to advise on potential causes given the comlexity of the project.

Try to isolate the problem by creating simpler test sketches.

fungreenfox, 2023-07-30:
Finally got this to work. An issue is, that the library wants to make uint8_t variable, while you see it logicly made with int8_t, and you know it never goes into negative numbers. 
After a while, i followed the library for the si4703 instead. The lesser mess with weird assigned variables has made the startup more stabile.

closed 2023-07-30 by fungreenfox