Bodmer/TFT_eSPI #2710: Error during compile exemples with funcction alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc);

brightproject, 2023-06-17:
I used BlackPill stm32f401ccu6 and ILI9486 3.5" display.
[image]
When I try compile example `Colour_Wheel.ino` I catch error
```
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\Colour_Wheel.ino.cpp.o: in function `setup':
Colour_Wheel.ino.cpp:(.text.setup+0x30): undefined reference to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)'
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: Colour_Wheel.ino.cpp:(.text.setup+0x40): undefined reference to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)'
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: Colour_Wheel.ino.cpp:(.text.setup+0x50): undefined reference to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)'
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: Colour_Wheel.ino.cpp:(.text.setup+0x62): undefined reference to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)'
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: Colour_Wheel.ino.cpp:(.text.setup+0x74): undefined reference to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)'
c:/users/admin/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\Colour_Wheel.ino.cpp.o:Colour_Wheel.ino.cpp:(.text.setup+0x86): more undefined references to `TFT_eSPI::alphaBlend(unsigned char, unsigned short, unsigned short)' follow
collect2.exe: error: ld returned 1 exit status
exit status 1
Compilation error for Generic STM32F4 series board.
```

I found in `TFT_eSPI.cpp` code
```
/***************************************************************************************
** Function name:           alphaBlend
** Description:             Blend 16bit foreground and background
*************************************************************************************x*/
inline uint16_t TFT_eSPI::alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc)
{
  // Split out and blend 5 bit red and blue channels
  uint32_t rxb = bgc & 0xF81F;
  rxb += ((fgc & 0xF81F) - rxb) * (alpha >> 2) >> 6;
  // Split out and blend 6 bit green channel
  uint32_t xgx = bgc & 0x07E0;
  xgx += ((fgc & 0x07E0) - xgx) * alpha >> 8;
  // Recombine channels
  return (rxb & 0xF81F) | (xgx & 0x07E0);
}

/***************************************************************************************
** Function name:           alphaBlend
** Description:             Blend 16bit foreground and background with dither
*************************************************************************************x*/
uint16_t TFT_eSPI::alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc, uint8_t dither)
{
  if (dither) {
    int16_t alphaDither = (int16_t)alpha - dither + random(2*dither+1); // +/-4 randomised
    alpha = (uint8_t)alphaDither;
    if (alphaDither <  0) alpha = 0;
    if (alphaDither >255) alpha = 255;
  }

  return alphaBlend(alpha, fgc, bgc);
}
```
My code
```
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library
TFT_eSprite painter = TFT_eSprite(&tft);

void setup() {

  tft.init();
  tft.setRotation(1);
  //Serial.begin(115200); // For debug
  tft.fillScreen(TFT_BLACK);
}

void loop() {
create_triangle();
}

void create_triangle() {
	const uint16_t tempColor = tft.alphaBlend(TFT_YELLOW, TFT_MAROON, 120);
	
tft.fillTriangle(10, 10, 50, 50, 100, 100, tempColor);
painter.fillTriangle(10, 10, 50, 50, 100, 100, tempColor);

}
```

Maybe the problem is that I'm using a display ILI9486?
Or is the compilation error problem that I do not use the image output through `TFT_eSprite`?
`TFT_eSprite painter = TFT_eSprite(&tft);`
Or is stm32f4 not compatible with alphablend?

Bodmer, 2023-06-17:
Parameter order is incorrect, try this:
tft.alphaBlend(120, TFT_YELLOW, TFT_MAROON);

brightproject, 2023-06-18:
Example from `Colour_Wheel.ino` have error also...
[image]

brightproject, 2023-06-19:
I checked file of [Processors](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.c)
I am using a stm32f401 with a display* connected via an 80 bit parallel interface.
Is it possible that the compilation error for the black pill is due to the fact that I changed the default pin numbers
```
 #define TFT_CS   PB8  // Chip select control pin
 #define TFT_DC   PB7  // RS/CD - Data Command control pin
 #define TFT_RST  PB9  // Reset pin
 #define TFT_WR   PB6  // Write strobe control pin 
 #define TFT_RD   PB0  // Read pin

#define TFT_D0   PA0  // 8 bit parallel bus to TFT
#define TFT_D1   PA1
#define TFT_D2   PA2
#define TFT_D3   PA3
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
```
to my choice?

```
#define TFT_CS   PB6  // Chip select control pin
#define TFT_DC   PB5  // RS/CD - Data Command control pin
#define TFT_RST  -1  // Reset pin
#define TFT_WR   PB1  // Write strobe control pin 
#define TFT_RD   PB0  // Read pin

#define TFT_D0   PA0  // 8 bit parallel bus to TFT
#define TFT_D1   PA1
#define TFT_D2   PA2
#define TFT_D3   PA3
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
```
Could this be a problem?
I can’t change the pins - the printed circuit board is so divorced.

brightproject, 2023-06-20:
An amazing thing, as soon as I removed from code
```
//inline uint16_t TFT_eSPI::alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc)
uint16_t TFT_eSPI::alphaBlend(uint8_t alpha, uint16_t fgc, uint16_t bgc)
{
  // Split out and blend 5 bit red and blue channels
  uint32_t rxb = bgc & 0xF81F;
  rxb += ((fgc & 0xF81F) - rxb) * (alpha >> 2) >> 6;
  // Split out and blend 6 bit green channel
  uint32_t xgx = bgc & 0x07E0;
  xgx += ((fgc & 0x07E0) - xgx) * alpha >> 8;
  // Recombine channels
  return (rxb & 0xF81F) | (xgx & 0x07E0);
}
```
word `inline `from the function, the code immediately began to compile without errors for the black pill STM32F401CCU6
[image]
I turned to the [GPT](https://gpt4.gravityengine.cc/) chat for help and he answered that inline is used to speed up the code, and it allegedly does not compile, but is placed in the compilation code as is.
Interestingly, the ESP32 normally "eats" inline, and the STM32F401CCU6 does not want to work with it.

Bodmer, 2023-06-30:
OK, it looks like the STM32 compiler does not like inline library functions if called directly by the sketch. Other functions in the library are inline but there arenot called from a sketch.

I will update the library. Thanks for finding the root cause.

Bodmer, 2023-07-05:
Librarymaster has been updated.

closed 2023-07-05 by Bodmer

UnexpectedMaker, 2023-11-05:
> Librarymaster has been updated.

Just posting that this issue also exists on the ESP32-S3 at least with ESP32 Arduino Core 3.0.0 alpha 2 - I commented out the `inline` in .h and .cpp files and it now compiles and works correctly.

```
#if !defined (STM32) && !defined(__IMXRT1052__) && !defined(__IMXRT1062__)
//   inline
#endif
```