Bodmer/TFT_eSPI #2718: ST7735 wrong color

demenkovms, 2023-06-21:
Hello. I use TFT_eSPI with my ESP32 in Arduino IDE, I tried display bmp but get not bright colors.

on display:
[image]

in original:
[image]
[image]

I thought that trouble was in img convert, but I try jpg from example and get the same result:

[image]

I use ST7735 display [(https://www.aliexpress.com/item/1005004565692996.html?spm=a2g0o.order_list.order_list_main.162.6e921802MyEB6N)](https://www.aliexpress.com/item/1005004565692996.html?spm=a2g0o.order_list.order_list_main.162.6e921802MyEB6N)

[image]

and my Setup.h:
```
#define ST7735_DRIVER

#define ST7735_REDTAB

#define TFT_WIDTH  320
#define TFT_HEIGHT 240

#define TFT_RGB_ORDER TFT_RGB

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000
```

Can you help what is wrong? Thanks!

Bodmer, 2023-06-30:
That display is not an ST7735 type. The vendors page title is only correct for one of the listed display options. It looks like you have selected and bought the ILI9341 display, check which one you bought, try the ILI9341 driver.

Bodmer, 2023-06-30:
Also try ST7789 driver.

Bodmer, 2023-07-28:
Assumed resolved.

closed 2023-07-28 by Bodmer