Bodmer/TFT_eSPI #2731: Strange display behavior - the image is tripled on ESP32

brightproject, 2023-07-05:
I am engaged in the adaptation of one open [project](https://www.huvver.tech/huvver-avi/) of aircraft instruments.
Initially, this code was written under `M5Stack`.
[image]
This is a complete device that has its own library, in which the `display.h` class is written and the `TFTe_SPI` library is initialized in it.
Moreover, `M5Stack` uses a far from new version of the `TFTe_SPI` library and programmers made excerpts from `TFTe_SPI`, only the most necessary for `M5Stack` to work.
I have my own hardware on the device.
[image]
I use a display with a ST7789 driver and an 8-bit parallel connection.
I use the following ESP32 contacts, I prescribe everything in the settings file of the `TFTe_SPI` library.
```
#define ESP32_PARALLEL

// #define ST7789_DRIVER
#define ST7789_2_DRIVER
#define TFT_INVERSION_OFF
// Screen

// ESP32 pins used
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14
#define TFT_BL   5
```
I wrote my own class for working with the display and connected the `TFTe_SPI` library as is using define
`#include <TFT_eSPI.h>`
I have two open source versions from HUVVER developers
I will call them OLD and NEW.
The OLD version is connected to the `TFTe_SPI` library using my class and works fine.
[image]
The NEW version only works on displaying the logo and then miracles begin.
[image]
Example code
```
TFT_eSprite gdraw = TFT_eSprite(&VVLcd);

  VVbegin();//  tft.begin(); tft.setRotation(1); tft.fillScreen(0);

  //
  // Display splash screen
  //
  VVLcd.setSwapBytes(true);
  VVLcd.fillScreen(TFT_BLACK);

  VVLcd.pushImage(0, 0, logoWidth, logoHeight, Logo);
```
The code functions normally inside the E`SP32` microcontroller, because there is an output of debugging information to the serial port.
The image is not displayed until I touch the G16 pin with my hand / finger, this is according to the settings
`#define TFT_D5   16`
When I touch `pin 16` with my finger, such a picture appears on the display ... which, of course, can hardly be called a normal image.
[image]
What could be the reason for such a strange display?
SetSwapByte is set the same in the OLD and NEW version of the code examples.
Pin number `16` is not used anywhere else except for `TFT_D5`.
Also, I still can’t understand the difference between the library used in `[M5Stack](https://github.com/m5stack/M5Stack/blob/master/src/utility/In_eSPI.h)` and the [TFTe_SPI](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h) library.
Older version 1[.4.20](https://github.com/Bodmer/TFT_eSPI/releases/tag/1.4.20)

brightproject, 2023-07-05:
I think I found the cause of my troubles.
In the sketch code, section SETUP, there is a code, serial2 port settings.
```
Serial2.setRxBufferSize (RXBUFFER);     
Serial2.setTxBufferSize (TXBUFFER);	  
Serial2.begin (BITRATE, SERIAL_8N1, PIN_RX2, PIN_TX2, RS232);
```
In the pin settings, I specified the pin values as -1.
```
// #define PIN_TX2 	17			// Serial 2 - default pin
// #define PIN_RX2 	16  		// Serial 2 - default pin
#define PIN_TX2 	-1			// Serial 2
#define PIN_RX2 	-1  		// Serial 2
```
Apparently, when you do not specify pins for the serial port, it sets them by default, and these are just pins numbered 16 and 17.
In general, the lack of pins is my eternal problem ...
I want to use the display at maximum speed, so I always connect via an 8-bit parallel interface, but more than half of the microcontroller pins are immediately occupied.
And difficulties begin with connecting buttons or encoders, as well as with peripherals in the form of a UART or CAN bus.
Advise how to redistribute the display connection so that there are no problems with the I / O pin registers on the ESP32?
And at the same time use peripherals without remapping.

Bodmer, 2023-07-05:
Great, glad you have solved it.

closed 2023-07-05 by Bodmer