Bodmer/TFT_eSPI #2739: drawWedgeLine call to base display (tft.) locks it up and stops tft.fillCircle

MikeyMoMo, 2023-07-10:
Using ESP32 WROOM.  No PSRAM for large sprites so I have to use the base display.  About to upgrade to WRover to get PSRAM.

I have a project that needs to draw two wedge lines onto the display.  I was using fillTriangle but it was not as pretty as WedgeLine so I switched over.  Suddenly things started locking up in one of 3 ways.  Either some dots did not get written, all dots got written or no dots got written.  It would change with any change to the code.  

I thought that maybe it was hardware failing.  I was in the middle of a large conversion to add a new display with a different touch controller on it so I though it might be something I had done.  So it took a lot of troubleshooting to pin it down.  I thought that maybe the integers going in might be confusing the routine although the compiler should be taking care of the conversion so I added (float) and float(myHX).  Neither helped.  No error messages, just stops updating as soon as the drawWedgeLine is issued.  The other board is a WT32-SC01 and has PSRAM so I don't call tft.drawWedgeLine and it runs just fine.  Again, making me suspect hardware.

Fails and stops further screen updates:
`tft.drawWedgeLine(myHX, myHY, (float)xCenter, (float)yCenter, 2., 8., TFT_RED);`

This also stops working after the drawWedge:
`tft.fillCircle(mySX, mySY, 3, ILI9341_YELLOW);`

Works:
`sprite61.drawWedgeLine(0, 0, 100, 50, 2., 8., TFT_WHITE);`

Bodmer, 2023-07-10:
These symptoms suggest lack of available memory which leads to corruption of variables and arrays. Check the size of memory resources consumed by the sprites. Bear in mind that the memory can get fragmented if sprites are often created and deleted.

Try a simpler sketch to test the drawing function works as expected. If it does not then post a simple sketch that demonstrates the problem.

MikeyMoMo, 2023-07-11:
Remember, when the same program writes the hands to a sprite, it works just fine.  Only when writing to the tft. object does it fail.

Sketch uses 1239285 bytes (39%) of program storage space. Maximum is 3145728 bytes.
Global variables use 50660 bytes (15%) of dynamic memory, leaving 277020 bytes for local variables. Maximum is 327680 bytes.

Just before the call to do the wedge hands: 
Current Heap Size: 108824
uxTaskGetStackHighWaterMark: 5404
Current Heap Size: 109052
uxTaskGetStackHighWaterMark: 5228
Current Heap Size: 114368
uxTaskGetStackHighWaterMark: 5228

I will try to produce a smaller version of the program soon.

What other numbers would you like to see to help with this?

#if defined DO_WEDGE_HANDS
  Serial.printf("Current Heap Size: %i\r\n",  esp_get_free_heap_size());
  Serial.print("uxTaskGetStackHighWaterMark: ");
  Serial.println(uxTaskGetStackHighWaterMark(NULL));
  tft.drawWedgeLine(myHX, myHY, xCenter, yCenter, 2, 8, TFT_RED);
  tft.drawWedgeLine(myMX, myMY, xCenter, yCenter, 2, 8, TFT_WHITE);

Bodmer, 2023-07-13:
Usually I find that if the fault signature changes when unrelated code is changed, then memory corruption is the cause. Another possibility is that the new touch library operates using interrupts and is locking up the SPI bus. Without a detailed knowledge of the code and operating environment I can only guess at the cause.

Please provide simple example code that demonstrates the problem.

Bodmer, 2023-07-28:
Assumed solved.

closed 2023-07-28 by Bodmer

MikeyMoMo, 2023-08-20:
We were down for 3 weeks because i6f a hurricane.  Nothing has changed.
Code still broken. I will look to see if there is some code change but I
library update seen.