Bodmer/TFT_eSPI #2740: ILI9163 with RP2040: wrong size and Y positions

mobil750, 2023-07-10:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur

    my display is a Joy-it SBC-LCD02 with ILI9163 controller; it is defined as a 128x128 pixel display. Setting HEIGHT to 128 only 3/4 of the screen is used. The Rest is scratch and the first line seems to be above the screen. Using HEIGHT 160, it is fine.
    To print correctly with e.g. font2 I have to set the Y position to 32, otherwise the first line(s) are invisible.
    
3. IDE (e.g. Arduino or PlatformIO) 
    same on both IDEs
    
4. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
    2.5.31
    
5. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
   Raspberry Pi Pico/RP2040 2.7.1
   
6. Procesor, e.g RP2040, ESP32 S3 etc
   RP2040
   
7. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
    ILI9163 included in package
    
8. Interface type (SPI or parallel)
    SPI
    
Plus further information as appropriate to the problem:

I used the example sketch delivered with the package "Test and Diagnostics\Colour_Test

1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

DO NOT paste code directly into the issue. To correctly format code put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies. [See here:](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

Example output:

```
  Serial.begin(115200);
  tft.init();
```

Bodmer, 2023-07-13:
This problem occurs because there is an offset of the 128x128 display zone within the maximum resolution area of the driver chip (128x160). Typically this offset is 32.

The display in this case has a different shift of 32 in rotation 0. The offsets are set here for rotation 0:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9163_Rotation.h#L14

If you do not want to modify the library, then the other option is to define the display as full resolution and use the setViewport function to compensate for the offset.

mobil750, 2023-07-24:
OK thanks for the hint. I think I will use the viewport function to be independent from lib updates.

closed 2023-07-28 by Bodmer