Bodmer/TFT_eSPI #2746: Remove all references to tft_setup.h and Sketch_with_tft_setup from code

MartyMacGyver, 2023-07-16:
It appears at some point the capability to include a `tft_setup.h` file (versus tampering with the library and its directory tree) was deprecated. If that's final, then please remove all the remaining references to `tft_setup.h` and `Sketch_with_tft_setup` from the codebase. Otherwise people will continue to spend significant time attempting to do something that's actually impossible with this library as it is today, based on all the references out there to this absent capability.

tochinet, 2023-10-25:
Maybe there is more discussion on this elsewhere, but I was always annoyed by the very idea of a requirement to "tamper with the library and its directory tree" to use it. Unless my stupidity makes me unable to understand how it should work, that breaks the rational of things like github and the updating of libraries. Since I didn't want to go that way, I copied the setup to my own sketch directory and added "#define USER_SETUP_LOADED // Indicates a specific setup is already loaded" to make it work. Much cleaner.

AndreyFly007, 2023-12-17:
I completely agree. It took 2 weeks to try to launch an ESP32S3 with a GC9A01 screen. the screen refused to work. I tried to enter #define USE_HSPI_PORT, USE_VSPI_PORT, and manually entered #define MISO 11 MOSI 11 CS10, but nothing helped.
When I started to look deeper and downloaded the example "Test and diagnostic/read user setup", I discovered that
that my SPI pins do not change and point to pin 23. There is no such pin on my board, and there is no SPI bus on this pin. I had to edit the pin number in the library. It is much simpler, clearer and more logical to indicate this pin in your code, passing its value to the library.

Полностью согласен. 2 недели ушло на попытку запустить ESP32S3 с экраном GC9A01. экран отказывался работать. Пробовал прописывать #define USE_HSPI_PORT, USE_VSPI_PORT, вручную прописывал #define MISO 11 MOSI 11 CS10, но ничего не помогало.
Когда начал разбираться глубже, загрузил пример "Test and diagnostic/read user setup", я обнаружил, что у меня SPI пины не меняются и указывают на 23 пин. Такого пина на моей плате нет, и SPI шины на этом пине нет. Пришлось в библиотеке править номер пина. Гораздо проще, понятнее и логичнее указать этот пин в своем коде, передав библиотеке его значение.

AndreyFly007, 2023-12-17:
Changes had to be made to TFT_eSPI_ESP32_S3.h
This is the only way I got the display to work

Изменения пришлось вносить в TFT_eSPI_ESP32_S3.h
Только так, я заставил работать дисплей

```
////////////////////////////////////////////////////////////////////////////////////////
// Make sure SPI default pins are assigned if not specified by user or set to -1
////////////////////////////////////////////////////////////////////////////////////////
#if !defined (TFT_PARALLEL_8_BIT)

  #ifdef USE_HSPI_PORT

    #ifndef TFT_MISO
      #define TFT_MISO -1
    #endif

    #ifndef TFT_MOSI
      #define TFT_MOSI 11
    #endif
    #if (TFT_MOSI == -1)
      #undef TFT_MOSI
      #define TFT_MOSI 11
    #endif

    #ifndef TFT_SCLK
      #define TFT_SCLK 12
    #endif
    #if (TFT_SCLK == -1)
      #undef TFT_SCLK
      #define TFT_SCLK 12
    #endif

  #else // VSPI port

    #ifndef TFT_MISO
      #define TFT_MISO -1
    #endif

    #ifndef TFT_MOSI
      #define TFT_MOSI 11
    #endif
    #if (TFT_MOSI == -1)
      #undef TFT_MOSI
      #define TFT_MOSI 11
    #endif

    #ifndef TFT_SCLK
      #define TFT_SCLK 12
    #endif
    #if (TFT_SCLK == -1)
      #undef TFT_SCLK
      #define TFT_SCLK 12
    #endif

    #if defined(CONFIG_IDF_TARGET_ESP32S3) || defined(CONFIG_IDF_TARGET_ESP32S2)
      #if (TFT_MISO == -1)
        #undef TFT_MISO
        #define TFT_MISO TFT_MOSI
      #endif
    #endif

  #endif

#endif
```