Bodmer/TFT_eSPI #2755: Characters missing from some fonts

WombatHollow, 2023-07-22:
When compiling using PlatformIO on VSCode for a TTGO T-Display board (ESP32-DOWDQ6 V3), the character + does not print for some font sized.  In place there appears to be a blank character although size not always consistant.

In this Example I print +-   the - alway prints but when font size is 6, 7 or 8 there is only black to left of -.   If I use font sizes 1, 2 or 4, then +- is printed in appropiate size font.

The font related Build Flags used in PlatformIO.ini are
 ```
 -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1  

```
Also from Platformio.ini
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

lib_deps =
    bodmer/TFT_eSPI   
    Button2@1.0.0

`
uses latest TFT_eSPI v 2.5.31

THe statement in main.cpp being used is

`    tft.drawString("+-", tft.width()-100, tft.height()-100, 1);   \\ where the last digit is 1,2,4,6,7 or 8 for different size fonts
`
this prints the characters in roughly middle of the display for purpose of testing

FYI TTGO T-Display uses ST7789V chip driven over SPI.   Not sure which TFT_eSPI driver is being used but if platformIO.ini is not overwriting, then it is the default as I have not modified either so appears to be using #define ILI9341_DRIVER in User_Setup.h

I did then change User_Setup_Select.h to use 
`#include <User_Setups/Setup25_TTGO_T_Display.h>`
and the 4 size font prints, but still no 6, 7 or 8 size font printing + character

[files.zip](https://github.com/Bodmer/TFT_eSPI/files/12135981/files.zip)

WombatHollow, 2023-07-22:
OK, I just discovered font 6 and larger are only intended to display numbers and Time aka 0 to 9 plus - and :
So I will close or delete

closed 2023-07-22 by Bodmer

DrJaymz, 2024-03-11:
Duh... This took me a couple of hours to work out!!!