Bodmer/TFT_eSPI #2763: drawRoundRect not displaying correctly on TFT 3.2 inch (220x340) ILI9341 driver.   Incorrect width (horizontal pixels)

jramongit, 2023-07-26:
drawRoundRect method/function not displaying rectangle with the right width.
I am using an ESP32 Dev Module connected though SPI to a 3.2 inch TFT which is 240x320 pixels and has a ILI9341 driver.

I call drawRoundRect to display a rounded rectangle around the full screen (320 pixels wide and 240 pixels height), but width is not displaying correctly; it is about 25% (1/4 th) of what is expected.

Attaching the code I'm using in Arduino IDE.

[TFT_eSPI_issue_drawRoundRect.zip](https://github.com/Bodmer/TFT_eSPI/files/12176805/TFT_eSPI_issue_drawRoundRect.zip)

[TFT drawRoundRect issue.zip](https://github.com/Bodmer/TFT_eSPI/files/12176848/TFT.drawRoundRect.issue.zip)

Bodmer, 2023-07-26:
The problem is in the sketch.

uint8_t maxX = 320;

This will not work, an 8 bit unsigned variable  has a maximum value of 255. A value of 320 will be truncated to 320 - 256 = 64

closed 2023-07-26 by Bodmer

jramongit, 2023-07-27:
My bad, I did the data type change and it worked.
Thanks a lot @Bodmer for responding so quickly.   I appreciate it.