Bodmer/TFT_eSPI #2764: Colours have been changing between TFT_eSPI V 2.4.72 and V 2.4.79

WillyIoBrok, 2023-07-26:
Using the colour "TFT_BLUE" gave with all older TFT_eSPI versions up to V 2.4.72 the colour blue in the display.
Since the release 2.4.79 (also with the newest 2.5.0) I am getting the colour red when selecting "TFT_BLUE"

The same happened to "TFT_YELLOW" which gives now lightblue and "TFT_SKYBLUE" which gives now grey. Maybe more colour selections are affected which I don't use.

I didn't do anythng but recompile after linstalling the new library version. 
Re-installing the version 2.4.72 gives again the correct colours!

-> So it's pretty easy to to reproduce.

2. IDE: Arduino with ESP32 extensions.
5. 2.4.71, 2.4.79, 2.5.0
7. ESP32
9. ST7735_DRIVER
10. SPI

Hope this info is sufficient...

[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/12176886/User_Setup.zip)

Bodmer, 2023-07-28:
I don't rememeber any changes that would do this. Try using one of the following options in the setup file:

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L76-L77

WillyIoBrok, 2023-07-29:
Bingo ! Your hint made my project working again also under TFT_eSPI V 2.5.0!

uncommenting the first "#define ..." gave my display the right colours back!

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

so what do I learn from that?

Since version 2.4.79 you MUST uncomment this line since some defaults seem to have changed after V 2.4.72
In older version you CAN uncomment this line - it doesn't do any harm. (I have tested this with 2.4.72)

Thanks a lot. Bodmer, for the excellent library and the quick response !

closed 2023-07-29 by WillyIoBrok