Bodmer/TFT_eSPI #2773: ili9486 color error.

acer1204, 2023-07-30:
i use user_setup.h to config ili9486 in this tft lcd.
http://www.lcdwiki.com/4.0inch_SPI_Module_ILI9486
the color has some problem.
how to fix it?
It looks like 16bit 18bit problem?
but when i define RPI_DISPLAY_TYPE, the tftlcde will show a white screen.
[image]

Bodmer, 2023-07-30:
Do not use RPI_DISPLAY_TYPE, it will not work.

Run the colour test example to see if red and blue are swapped.

acer1204, 2023-08-02:
> Do not use RPI_DISPLAY_TYPE, it will not work.
> 
> Run the colour test example to see if red and blue are swapped.

Which example is the color test?

Bodmer, 2023-08-02:
It is an example, see [here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Colour_Test).

Bodmer, 2023-08-14:
Assumed resolved.

closed 2023-08-14 by Bodmer