Bodmer/TFT_eSPI #2775: Hii iam try to pair esp32 to ILI9486 8bit parallel display with touch but iam facing this error =    Touch functions not supported in 8/16 bit parallel mode or with RP2040 PIO.        #error >>>>------>> Touch functions not supported in 8/16 bit parallel mode or with RP2040 PIO

DhirajDSD, 2023-08-01:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
2. IDE (e.g. Arduino or PlatformIO)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
5. Procesor, e.g RP2040, ESP32 S3 etc
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
7. Interface type (SPI or parallel)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

DO NOT paste code directly into the issue. To correctly format code put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies. [See here:](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

Example output:

```
  Serial.begin(115200);
  tft.init();
```

Bodmer, 2023-08-02:
The library does not support touch functions in parallel mode.

If you have a SPI, I2C or resistance based touch screen then another library will be required. The touch library selected must support the touch controller fitted and the touch controller interface (SPI, I2C or resistance measurement).

closed 2023-08-02 by Bodmer

DhirajDSD, 2023-08-04:
How can we include seperate library for 8-bit parallel

Bodmer, 2023-08-04:
Yes, use a separate library. You are likely to find an Arduino library for your touch controller with a Google search.

LazaroFilm, 2023-08-05:
How do we make the external library communicate with TFT_eSPI? is are there certain function calls or expected returns necessary?

Bodmer, 2023-08-06:
No library linking is required.  Checking for touch and the drawing the graphics are all done in the sketch.

DhirajDSD, 2023-08-10:
hii bodmer how to link <TouchScreen.h> to <TFT_eSPI.h> if any example code available please share me

LazaroFilm, 2023-08-10:
I realized that tft_eSPI actually doesn't need any touchpad input. You would write that in your code and update your display accordingly. If you want your GUI to be reactive Yoo can try LVGL. It creates widgets you can use and apply. LVGL does have a function it ads you it's boilerplate to read your touchscreen inputs.

DhirajDSD, 2023-08-16:
> No library linking is required. Checking for touch and the drawing the graphics are all done in the sketch.

Thanks Bodmer, I have successfully linked together "Touch Screen.h" and <TFT_eSPI.h> and Touch is working properly

DhirajDSD, 2023-08-16:
but it is possible to work touch on squareline studio using  "Touch Screen.h" and <TFT_eSPI.h>

Bodmer, 2023-08-16:
I am not familiar with squareline studio so am unable to help.

morsteevo, 2024-02-10:
can you share your code with how you got this working? please