Bodmer/TFT_eSPI #2782: MD144_SPI 128x128 display disturbed

MHz000, 2023-08-08:
the very inexpensive display MD144_SPI 128x128 with ST7735s cip compiles error-free with the Setup7_st7735_128x128.h file. 
I am using an ESP8266 processor with the following connections:
SCLK on D5, MOSI on D7, CS on D8,DC on D3 and RST via 4.7k -> 3.3V. 
Unfortunately, the representation on the display is disturbed. This becomes clear in displaying  TFT_Ellipse example.
I would be grateful for a hint what I need to correct, be it in the setup, in the program or the wiring. Thank you
[image]
[image]

MHz000, 2023-08-09:
I tried everything that came to my mind as a not really an experienced C++ programmer:
. use the ESP32 instead of the ESP8266
. reduce the SPI speed 
. inserted a delay after the tft.init()
. tested different pin assignments
All of this was unsuccessful. Rudimentary ellipses appear on the display, but they are very disturbed. So I think the display itself and the processor are ok, but something is wrong in the setup.
I would be grateful for any hint as to what else needs to be done.

What I did:
  SPI.beginTransaction(SPISettings(3500000, MSBFIRST, SPI_MODE0));
  tft.init();
  delay(200);
  tft.setRotation(1); ...

my pin assignments:
  // ESP8266
  #define USER_SETUP_ID 7
  #define TFT_MOSI PIN_D7   
  #define TFT_SCK  PIN_D5   
  #define TFT_CS   PIN_D8   
  #define TFT_DC   PIN_D3    
  //#define TFT_RST  PIN_D4    
  #define TFT_RST  -1   

  //ESP32
  #define USER_SETUP_ID 7
  #define TFT_SCK  PIN_D18    
  #define TFT_MOSI PIN_D23    
  #define TFT_CS   PIN_D5     
  #define TFT_DC   PIN_D19    
  #define TFT_RST  -1

Bodmer, 2023-08-14:
What power voltage are you conecting to the display?

The SPI speed is set by the library so setting is a sketch will not work. What SPI clock rate are you using in the setup file?

MHz000, 2023-08-15:
Thanks for your answer. 
Even if I use a 10 amp external power supply nothing changes. I also have a 1000µ capacitor parallel to the power rails. 
As you say, nothing changes if I change the SPI speed in the program, for example with SPI.beginTransaction(SPISettings(xxx, MSBFIRST, SPI_MODE0)); Is there another way to lower this. It's a very cheap display, maybe that helps. 
Because as you can see in the first picture, something similar to an ellipse is displayed. By the way, this effect appears on two different displays of the same type.
Thank you for your great library, which I use successfully in other projects with other displays.

MHz000, 2023-08-15:
just to be sure, here is the program i use for the display test: Sorry if
I made a stupid mistake copying the example 😞
using VScode and PlatformIO

Bodmer, 2023-08-15:
To rephraze the question. Is the display powered from5V or 3.3V in your setup?

MHz000, 2023-08-16:
many thanks for your response. Originally I had supplied the display with the 3.2 V voltage from the ESP. This is when the disturbances occurred. I changed the jumpers on the display to 5V supply and the display now shows a stable white background.
I forgot to mention that the following messages appear when compiling. As with other programs/libraries, I ignored these with no consequences.
Thank you for your support and your efforts

2 times: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]

#warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]

closed 2023-08-16 by Bodmer

MHz000, 2023-08-17:
thanks for your efforts Bodmer