Bodmer/TFT_eSPI #2788: TFT and WiFi conflict

JeanVictor100, 2023-08-15:
When I run the code below, for some reason the screen is completely buggy after using the wifi, it stops responding to drawing commands, and for some reason my esp32 keeps the onboard led blinking non-stop. By the way, I'm using an esp32 and a TFT screen 3.2" 320 x 240 ILI9341 with the TFT_eSPI 2.5.31 library.
```
#include "FS.h"  // Touch Calibration
#include <SPI.h>
#include <TFT_eSPI.h>  // Hardware TFT
// WiFi
#include <WiFi.h>
//#include <HTTPClient.h>

TFT_eSPI tft = TFT_eSPI();  // TFT custom library

#define CALIBRATION_FILE "/TouchCoordData"

const char* WIFI_SSID = "ssid_here";
const char* WIFI_PASSWORD = "pass_here";

void setup() {
  Serial.begin(9600);

  welcome();

  delay(5000);

  myScreen();
}

void loop() {
  uint16_t x, y;

  // Verify touch
  if (tft.getTouch(&x, &y)) {
    tft.fillCircle(x, y, 2, TFT_WHITE);
  }
}

// Touch Calibration
void touch_calibrate() {
  uint16_t calData[5];
  uint8_t calDataOK = 0;

  // check file system exists
  if (!SPIFFS.begin()) {
    Serial.println("Formatting file system");
    SPIFFS.format();
    SPIFFS.begin();
  }

  // check if calibration file exists and size is correct
  if (SPIFFS.exists(CALIBRATION_FILE)) {
    File f = SPIFFS.open(CALIBRATION_FILE, "r");
    if (f) {
      if (f.readBytes((char*)calData, 14) == 14)
        calDataOK = 1;
      f.close();
    }
  }

  // Checks if the touch calibration is valid
  if (calDataOK) tft.setTouch(calData);
  else {
    // Touch error
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(20, 0);
    tft.setTextFont(2);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.println("Touch Error!");
  }
}

// Display the home screen
void welcome() {
  tft.init();
  tft.setRotation(0);  // vertical
  touch_calibrate();   // calibration
  tft.fillScreen(TFT_BLACK);

  tft.setTextFont(4);
  tft.setTextSize(1);
  tft.setTextColor(TFT_CYAN, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome!");
}

void myScreen() {
  // WiFi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  /*
  --- Does something, like an http connection in my case
  */

  tft.fillScreen(TFT_BLACK);
  // Text
  tft.setTextFont(2);
  tft.setTextSize(1);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome again!");
}
```

I tried to disconnect esp32 from wifi after finalizing the necessary transactions, so I added the code "WiFi.disconnect();" after the comment "Does something, like an http connection in my case" but the display still fails.

Continuing the tests I realized that the display works normally if I do the following:

```
#include "FS.h"  // Touch Calibration
#include <SPI.h>
#include <TFT_eSPI.h>  // Hardware TFT
// WiFi
#include <WiFi.h>
//#include <HTTPClient.h>

TFT_eSPI tft = TFT_eSPI();  // TFT custom library

#define CALIBRATION_FILE "/TouchCoordData"

const char* WIFI_SSID = "ssid_here";
const char* WIFI_PASSWORD = "pass_here";

void setup() {
  Serial.begin(9600);

  //welcome();

  delay(5000);

  myScreen();
}

void loop() {
  uint16_t x, y;

  // Verify touch
  if (tft.getTouch(&x, &y)) {
    tft.fillCircle(x, y, 2, TFT_WHITE);
  }
}

// Touch Calibration
void touch_calibrate() {
  uint16_t calData[5];
  uint8_t calDataOK = 0;

  // check file system exists
  if (!SPIFFS.begin()) {
    Serial.println("Formatting file system");
    SPIFFS.format();
    SPIFFS.begin();
  }

  // check if calibration file exists and size is correct
  if (SPIFFS.exists(CALIBRATION_FILE)) {
    File f = SPIFFS.open(CALIBRATION_FILE, "r");
    if (f) {
      if (f.readBytes((char*)calData, 14) == 14)
        calDataOK = 1;
      f.close();
    }
  }

  // Checks if the touch calibration is valid
  if (calDataOK) tft.setTouch(calData);
  else {
    // Touch error
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(20, 0);
    tft.setTextFont(2);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.println("Touch Error!");
  }
}

// Display the home screen
void welcome() {
  tft.init();
  tft.setRotation(0);  // vertical
  touch_calibrate();   // calibration
  tft.fillScreen(TFT_BLACK);

  tft.setTextFont(4);
  tft.setTextSize(1);
  tft.setTextColor(TFT_CYAN, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome!");
}

void myScreen() {
  // WiFi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  /*
  --- Does something, like an http connection in my case
  */
  WiFi.disconnect();

  tft.init();
  tft.setRotation(0);  // vertical
  touch_calibrate();   // calibration
  tft.fillScreen(TFT_BLACK);
  // Text
  tft.setTextFont(2);
  tft.setTextSize(1);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome again!");
}
```

That is, if I comment out the welcome method and initialize the screen after the wifi call, I would have no problems, but I need to display information on the screen, allow the user to use the wifi looking for more information to display on the screen again.

I don't know if I should post this issue on the bugs page or the discussions page, or if this bug has already been fixed or if I'm just talking rubbish, but whatever.

Bodmer, 2023-08-16:
Does the serial monitor output indicate the processor has crashed or is it still running around in loop()?

JeanVictor100, 2023-08-16:
All code after using wifi works normally, I can display messages normally on the serial monitor for example, and the same goes for the loop, it works normally. Case it helps, here is the "User_Setup" file configuration.

```
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
//#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

// Tell the library to use parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT
//#defined TFT_PARALLEL_16_BIT // **** 16 bit parallel ONLY for RP2040 processor ****

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
//#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1351_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER
//#define GC9A01_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
[... 292 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```

Bodmer, 2023-08-18:
The setup file looks fine.  Typically the GPIO 2 is connected to the onboard LED and that is used for the DC line so check your boards documents to see which pin the LE is on and avoid that for a TFT control pin if a flashing LED is not wanted.

As for the lockup of the screen. Typically this ocurs if the screen gets garbage commands, this can happen for a number of reasons:

Power glitches
Long wires
inadequate ground connections (too long, wire too thin)

I have not seen this behaviour myself and my projects do sometimes use WiFi so it seems peculiar to your hardware.

Try adding a tft,begin() again in the sketch somewhere to try and initiaise the display again and get it out of the locked up condition. You will then need to set the rotation again.

JeanVictor100, 2023-08-19:
So, i tried to add "tft.begin()" but didn't solve the problem.

My wires measure an average of 3 inches, but I can remake the prototype on another protoboard, trying to reduce the wire length. I'll do it during the week, and I'll come back with the result.

Bodmer, 2023-08-21:
3" should be fine so that is unlikely to be the problem.

JeanVictor100, 2023-08-23:
So, observe the screen's behavior after using Wi-Fi and the "disconnect()" function with the code below, then tell me, should this be happening or is it happening only to me?

```
#include "FS.h"  // Touch Calibration
#include <SPI.h>
#include <TFT_eSPI.h>  // Hardware TFT
// WiFi
#include <WiFi.h>
//#include <HTTPClient.h>

TFT_eSPI tft = TFT_eSPI();  // TFT custom library

#define CALIBRATION_FILE "/TouchCoordData"

const char* WIFI_SSID = "ssid_here";
const char* WIFI_PASSWORD = "pass_here";

void setup() {
  Serial.begin(9600);

  welcome();

  delay(5000);

  myScreen();
}

void loop() {
  uint16_t x, y;

  // Verify touch
  if (tft.getTouch(&x, &y)) {
    tft.fillCircle(x, y, 2, TFT_WHITE);
  }
}

// Touch Calibration
void touch_calibrate() {
  uint16_t calData[5];
  uint8_t calDataOK = 0;

  // check file system exists
  if (!SPIFFS.begin()) {
    Serial.println("Formatting file system");
    SPIFFS.format();
    SPIFFS.begin();
  }

  // check if calibration file exists and size is correct
  if (SPIFFS.exists(CALIBRATION_FILE)) {
    File f = SPIFFS.open(CALIBRATION_FILE, "r");
    if (f) {
      if (f.readBytes((char*)calData, 14) == 14)
        calDataOK = 1;
      f.close();
    }
  }

  // Checks if the touch calibration is valid
  if (calDataOK) tft.setTouch(calData);
  else {
    // Touch error
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(20, 0);
    tft.setTextFont(2);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.println("Touch Error!");
  }
}

// Display the home screen
void welcome() {
  tft.init();
  tft.setRotation(0);  // vertical
  touch_calibrate();   // calibration
  tft.fillScreen(TFT_BLACK);

  tft.setTextFont(4);
  tft.setTextSize(1);
  tft.setTextColor(TFT_CYAN, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome!");
}

void myScreen() {
  // WiFi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("Connecting");
  while(WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  /*
  --- Does something, like an http connection in my case
  */
  WiFi.disconnect();

  tft.fillScreen(TFT_BLACK);
  // Text
  tft.setTextFont(2);
  tft.setTextSize(1);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(60, 60);
  tft.println("Welcome again!");
}
```

Before:
[image]

After:
[image]

JeanVictor100, 2023-08-24:
But what I don't understand is that:
Using **wifi** and then the **display** = Okay
Using **display first**, then **wifi**, then **display again** = Display doesn't work correctly
I can't draw anything else on the screen or change the background color anymore.

That's what left me confused.

Bodmer, 2023-08-24:
I have run the last posted sketch above on two displays and it works fine.

What versions of TFT_eSPI and the ESP32 board package are you using?

JeanVictor100, 2023-08-24:
Bro... I feel very ashamed for such inattention... I had configured for another ESP and completely forgot about it. I've now changed it to the correct version, and it's works fine. I apologize for the time I made you lose.

closed 2023-08-24 by JeanVictor100