Bodmer/TFT_eSPI #2790: 4-Bit Sprite creation leads to crash

bkZuendorf, 2023-08-15:
```
  spr.setColorDepth(4);
  spr.createSprite(width, height);
```
leads to crash of ESP8266. Could solve the problem by moving the code line of method 
TFT_eSprite::createSprite(int16_t w, int16_t h, uint8_t frames)

```
  if ( (_bpp == 4) && (_colorMap == nullptr)) createPalette(default_4bit_palette);
```
to

```
  if (_img8)
  {
    _created = true;
    **if ( (_bpp == 4) && (_colorMap == nullptr)) createPalette(default_4bit_palette);**
    rotation = 0;
    setViewport(0, 0, _dwidth, _dheight);
    setPivot(_iwidth/2, _iheight/2);
    return _img8_1;
  }
```
the "createPalette" call in the current version won't create a palette due the variable _created isn't set! That's the reason I moved it after the "_created=true" statement.

Bodmer, 2023-08-15:
If a sprite is created then the palette can be created.   See example:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Sprite_image_4bit/Sprite_image_4bit.ino

Are you saying that example sketch fails?

bkZuendorf, 2023-08-16:
I didn't test with the sample file.
But I just had an simple file using standard syntax similar to:

```
TFT_eSPI    tft = TFT_eSPI();         // Declare object "tft"
TFT_eSprite spr = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

void create4BitSprite()
{
  spr.setColorDepth(4);
  spr.createSprite(WIDTH, HEIGHT);
  ...
}

void setup() {
  tft.init();
  create4BitSprite();
}
```
in the programm I create 16Bit and 4Bit versions depending on the situation dynamically.

I didn't find the exact place of the crash, but it seemed convincing to me, that a missing color table could be the problem.
The code
```
  if ( (_bpp == 4) && (_colorMap == nullptr)) createPalette(default_4bit_palette);
```
makes no sense to me, as the function call "createPalette" won't do anything as the _created flag is set to false.
So I moved the code after setting the flag and the crash disappeared!

Previously I checked whether the bitmap buffer has been allocated properly!

Bodmer, 2023-08-16:
Fixed in Github master

commit 5c1f163b (2023-08-16): Fix #2790
closed 2023-08-16 by Bodmer