Bodmer/TFT_eSPI #2796: TFT_eSPI::alphaBlend24 produce wrong mixing

aktawa, 2023-08-18:
The masking of the channels in the function is erroneous, i.e red and blue forground channels use the same bitmask as red

The function could be corrected as
```
uint32_t TFT_eSPI::alphaBlend24(uint8_t alpha, uint32_t fgc, uint32_t bgc, uint8_t dither)
{

  if (dither) {
    int16_t alphaDither = (int16_t)alpha - dither + random(2*dither+1); // +/-dither randomised
    alpha = (uint8_t)alphaDither;
    if (alphaDither <  0) alpha = 0;
    if (alphaDither >255) alpha = 255;
  }

  uint32_t rxx = bgc & 0xFF0000;
  rxx += ((fgc & 0xFF0000) - rxx) * alpha >> 8;
  uint32_t xgx = bgc & 0x00FF00;
  xgx += ((fgc & 0x00FF00) - xgx) * alpha >> 8;
  uint32_t xxb = bgc & 0x0000FF;
  xxb += ((fgc & 0x0000FF) - xxb) * alpha >> 8;
  return (rxx & 0xFF0000) | (xgx & 0x00FF00) | (xxb & 0x0000FF);
}
```

Bodmer, 2023-08-18:
I can't see any difference between the code above and the code here in the master:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L5011-L5028

I can only assume that this has been corrected as some point in the past.  What version of TFT_eSPI are you using?

closed 2023-08-30 by Bodmer

aktawa, 2023-08-31:
I am using the library version bodmer/TFT_eSPI@^2.5.31.

The bug is in the bitmask for the foreground (0xFF0000 vs 0x00FF00 for green, and 0xFF0000 vs 0x0000FF for blue). For example
`#include "TFT_eSPI.h"

void setup() {
  Serial.begin(115200);
  TFT_eSPI tft = TFT_eSPI(); 
  
Serial.println( tft.alphaBlend24(0x80u, 0x00888888u, 0x00000000u), HEX);
Serial.println( tft.alphaBlend24(0x80u, 0x00000088u, 0x00880000u), HEX);
Serial.println( tft.alphaBlend24(0x00u, 0x00224466u, 0x00664422u), HEX);
Serial.println( tft.alphaBlend24(0x80u, 0x00224466u, 0x00664422u), HEX);
Serial.println( tft.alphaBlend24(0xFFu, 0x00224466u, 0x00664422u), HEX);
}

void loop() {
  // put your main code here, to run repeatedly:
}`

returns
`444444
440044
660000
442233
22A965`

After applying the patch, the output is, (what i would expect)

`444444
440044
664422
444444
224465`

[alphaBlend24.patch](https://github.com/Bodmer/TFT_eSPI/files/12488675/alphaBlend24.patch)

Bodmer, 2023-09-01:
OK, understood. Fixed in master.

commit 0a477302 (2023-09-01): Fix #2796 (#2827)