Bodmer/TFT_eSPI #2803: Seeed XIAO RP2040 & ILI9341 Touch not working

dslocum, 2023-08-20:
I have a Seeed XIAO RP2040 with the above connections.  The display works fine with any of the demos I've tried, but I can't get any touch examples to run.

I'm currently using Test_Touch_Controller and the Serial monitor shows all 3 parameters = "0".  On a o'scope there are reasonable signals on the pins and I even get outputs on the MISO line that appear to have data changing as I move my finger / stylus.
[image]

Anything else I could look for?

_Originally posted by @dslocum in https://github.com/Bodmer/TFT_eSPI/discussions/2230#discussioncomment-6769564_

Bodmer, 2023-08-20:
I am guessing you are trying to use the PIO based SPI interface which does not support SPI reads. See here:

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup61_RP2040_ILI9341_PIO_SPI.h#L7-L11

Make sure line 11 is commented out.

Bodmer, 2023-08-30:
Asumed resolved.

closed 2023-08-30 by Bodmer