Bodmer/TFT_eSPI #2808: drawString() crashing

Dawid009, 2023-08-22:
I'm using TFT_eSPI on my esp32 board. OOP and pointer type "tft" so using  " -> " instead of " . ".
tft->fillRect, tft->drawRect, init, rotation... are ok, but "tft->drawString("text",100,100);" is crashing my app. I have no idea what is going on and why it is not working. The compiler does not detect errors but esp reboots when it is supposed to display text (other shapes are displayed).

Serial monitor: "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

Bodmer, 2023-08-24:
No idea what happened to the comment I posted!

I have used -> in support libraries so I see no reason why it should not work.

I think that something is going out of scope but it is not possible to determine this from the limited information.

Bodmer, 2023-08-24:
I tried this simple sketch and it works fine:
```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

TFT_eSPI *ptft = &tft;

void setup(void) {
  tft.init();
  tft.fillScreen(0);
  ptft->drawString("Hello World", 50, 50);
}

void loop() {
}
```

Dawid009, 2023-08-25:
TY for your reply. I noticed something was wrong with the fonts. If I add a fourth parameter to DrawString it works without a problem. setTextFont called before didn't work, maybe something was wrong with the scope in my code.

closed 2023-08-25 by Dawid009