Bodmer/TFT_eSPI #2820: sprintf didn't work with the type of double

ericzhang-debug, 2023-08-29:
Hello,
I tried the sprintf to format a temp[30] with a double number , but it only showed the string but no number.
```cpp
    char temp[100];
    double num=60.32;
    sprintf(temp, "Number : %3.2f", num);
    tft.fillScreen(TFT_BLACK);
    tft.print(temp);
```
It only shows: Number : 

How can I solve this problem?
Thank you very much !

Bodmer, 2023-08-30:
I suspect this may be a problem with the STM32 board package as it works fine on the ESP32. This is the test sketch:
```
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);
  tft.init();
  tft.fillScreen(TFT_BLACK);
  delay(5000); // Wait for Serial monitor window to be opened
}

void loop()
{
  char temp[100];
  double num = 60.32;
  sprintf(temp, "Number : %3.2f", num);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE);
  tft.setCursor(0, 0);
  tft.print(temp);
  Serial.print(temp);

  while (1) delay(100);
}
```
The following appears on the display and in the serial monitor window. Check the serial output in your software environment is correct:

Number : 60.32

Bodmer, 2023-08-30:
I just had a look in the IDE and it appears you have to chose the right runtime library to get printf:
[image]

closed 2023-08-30 by Bodmer