Bodmer/TFT_eSPI #2824: 'class TFT_eSPI' has no member named 'getTouch'

yoroboticscolombia, 2023-08-31:
Hi, Im using the library in arduino ide, however I get the error> 
'class TFT_eSPI' has no member named 'getTouch'
I uncommented the line #define TOUCH_CS 21 , in usersetup.h, any ideas why it keeps happening, 

board esp32.
driver ili9341.
display 2.8inch
I tried evertything, with no success. appreciate your help.

Bodmer, 2023-08-31:
Run the Read_User_Setup sketch and look at the serial output to verify the touch pin is listed.

yoroboticscolombia, 2023-08-31:
Hi, thanks, the touch pin is not listed, see: 

[image]

Bodmer, 2023-09-01:
Check User_Setup_Select.h is only calling up one setup file and it is the correct one.

yoroboticscolombia, 2023-09-02:
well, I was able to compile, by adding #define TOUCH_CS 21, inside every driver setup, and it worked, maybe you should update the library.  also, do you know if st7735 has lower resolution than ili9341?

I get it in the display, since ili9341, is cut, but i tried setting rotation and did not work, thanks.

[image]

Bodmer, 2023-09-02:
There is a problem somewhere in your setup path.

Adding #define TOUCH_CS 21 inside every driver setup is not needed or desirable normally.

Check User_Setup_Select.h is only calling up ONE setup file and it is the correct one.

Since you have it working I will close.

closed 2023-09-02 by Bodmer