Bodmer/TFT_eSPI #2829: Image incomplete and pixelation when using TFT_eSPI with ST7796 SD ESP32

Jc096, 2023-09-01:
Using the SPI TFT ST7796 screen and the TFT_eSPI library to project an image with the SD card, the image comes out incomplete and pixelated, the diode that generates the fault in that screen model has already been removed but the fault continues the same, I add photos and evidence of the error, I would greatly appreciate your help.
the configuration of the connection pins is the following:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC   4  // Data Command control pin
#define TFT_RST  17  // Reset pin (could connect to RST pin)
[image]
[image]

https://github.com/Bodmer/TFT_eSPI/assets/143748869/cd23cfc4-f1ab-4fcb-b99c-6d0c41fab537

Jc096, 2023-09-04:
here are the programming files:

[proyecto_prueba_demo.zip](https://github.com/Bodmer/TFT_eSPI/files/12515289/proyecto_prueba_demo.zip)

Bodmer, 2023-09-20:
Assumed resolved.

closed 2023-09-20 by Bodmer

Jc096, 2023-09-26:
Unresolved problem, the recommended code adjustments were made and the CS Pin was moved to GPIO Pin 22 and the operation remains the same, I attach evidence videos and the programming files, any comments that can help I will appreciate it

https://github.com/Bodmer/TFT_eSPI/assets/143748869/fe48bba0-0a92-4907-bfb8-2b1966a48746

https://github.com/Bodmer/TFT_eSPI/assets/143748869/6dae4dfd-aa51-4873-b388-9e3486a886a9

https://github.com/Bodmer/TFT_eSPI/assets/143748869/b168407e-bbfe-4bfa-b4e1-e6ea7444eff0