Bodmer/TFT_eSPI #2833: ILI9361 tft mirrored flip image

yoroboticscolombia, 2023-09-04:
HI, is there any way or register to invert ili9361, I get mirrored image, i have tried different rotation opcions but does not, its esp32, 

Thanks.

[image]

animevietsub, 2023-09-04:
Check if you are using `tft.setRotation(0);` try  `tft.setRotation(6);` which is the X-Mirror of (0)

yoroboticscolombia, 2023-09-05:
Thanks, the ili9361 has only rotation 3, in the options, is there anyway to update, 

// This is the command sequence that rotates the ILI9163 driver coordinate frame

  rotation = m % 4;

  writecommand(TFT_MADCTL);
  switch (rotation) {
    case 0:
      writedata(TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _init_width;
      _height = _init_height;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 0;
	  #endif
      break;
    case 1:
      writedata(TFT_MAD_MV | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _init_height;
      _height = _init_width;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 0;
	  #endif
      break;
    case 2:
      writedata(TFT_MAD_BGR);
      _width  = _init_width;
      _height = _init_height;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 32;
	  #endif
      break;
    case 3:
      writedata(TFT_MAD_MX | TFT_MAD_MV | TFT_MAD_BGR);
      _width  = _init_height;
      _height = _init_width;
	  #ifdef CGRAM_OFFSET
        colstart = 32;
        rowstart = 0;
	  #endif
      break;
  }

animevietsub, 2023-09-05:
`writedata(TFT_MAD_MX) `make the display Mirror-X, which case you are using ?

yoroboticscolombia, 2023-09-05:
> MX

thanks, I used all 3 cases, but no success,

animevietsub, 2023-09-05:
Are you using ILI9163 library?, you say you have ILI9361 chip right ? From the picture that should be the ILI9341 chip.

yoroboticscolombia, 2023-09-05:
> Are you using ILI9163 library?, you say you have ILI9361 chip right ? From the picture that should be the ILI9341 chip.

yes, its the only one that has good resolutuion, 9341 has blurry pixels.

animevietsub, 2023-09-05:
I have no idea why that happens, so at this time just try to edit the library for your case. Try to edit tft.setRotation(0) and remove the TFT_MAD_MX. 
```
 case 0:
      writedata(TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _init_width;
      _height = _init_height;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 0;
	  #endif
      break;
```

yoroboticscolombia, 2023-09-05:
> I have no idea why that happens, so at this time just try to edit the library for your case. Try to edit tft.setRotation(0) and remove the TFT_MAD_MX.
>
> ```
>  case 0:
>       writedata(TFT_MAD_MY | [...]

Thanks, I Will try tomorrow and tell you, since I did not understand the registers of the display in the datasheet,

Bodmer, 2023-09-06:
The library supports the ILI9163, I assume there is a type in your driver number?

Assuming that is the case then it the flip settings in the MADCTL register need to be changed by experiments. Once changed for rotation use a setRotation(x) command immediately after tft.init to make sure the setting overrides the default init.

The control bits MX, MY, MV flip the x,y and vertical coordinates.