Bodmer/TFT_eSPI #2834: Problem with Arduino-Nano-ESP32 and TFT-Round-GC9A01 with sda, scl, dc, cs

erikbartmann, 2023-09-05:
Hi,

I have some problems to get the TFT_eSPI-Sketches running.
I'm using a Arduino-Nano-ESP32 and TFT-Round-GC9A01 with sda, scl, dc, cs, rst.
I connected the following pins:

SDA from TFT to Nano-ESP32 Pin D11
SCL from TFT to Nano-ESP32 Pin D13
DC from TFT to Nano-ESP32 Pin D9
CS from TFT to Nano-ESP32 Pin 10
RST is not conected but I tried it to connect it to the RST-Pin of the nano-ESP32-Board :-(

I configured the User:Setup.h as shown:

`#define GC9A01_DRIVER
#define TFT_MOSI 11 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 13
#define TFT_CS   10  // Chip select control pin
#define TFT_DC    9  // Data Command control pin
//#define TFT_RST  33  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL   22  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
`
I tried the simple sketch
`
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup()
{
   tft.init();
   tft.fillScreen(TFT_BLACK);
   tft.setCursor(50, 50, 4);
   tft.setTextColor(TFT_WHITE, TFT_BLACK);
   tft.println("Hello, World!");
}

void loop() {}
`

Nothing happens on the TFT.

What's wrong here?

Best

Erik

Bodmer, 2023-09-06:
Run the Read_User_Setup sketch and look at the serial output to make sure the right pins are being picked up.

copyCATandpasteDOG, 2023-09-07:
Seems you did not uncomment the back light line : #define TFT_BL 22 // LED back-light

erikbartmann, 2023-09-08:
I do not have a BL-Signal on my display. It only have Vcc, GND, SCL, SDA, DC, CS, RST

erikbartmann, 2023-09-08:
Here is my picture...
[image]

erikbartmann, 2023-09-08:
> Run the Read_User_Setup sketch and look at the serial output to make sure the right pins are being picked up.
The result is:
TFT_eSPI ver = 2.5.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9A01
Display width  = 240
Display height = 320

MOSI    = GPIO 11
MISO    = GPIO 11
SCK     = GPIO 13
TFT_CS   = GPIO 10
TFT_DC   = GPIO 9

And in that way, I connetect everything. On my board there is no BL-pin. Maybe that's the problem? I put a picture in my last post here.

Thanks again!

Erik

copyCATandpasteDOG, 2023-09-08:
My bad,you're right,
i have the same as your's but forgot the pin details,i use it with a esp32 , my setup working:

```
//GC9A01 ROUND TFT & ESP32

#define GC9A01_DRIVER

#define TFT_MOSI 23 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 18
#define TFT_CS   22  // Chip select control pin
#define TFT_DC   16  // Data Command control pin
#define TFT_RST  4  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   -1  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

//#define SPI_FREQUENCY  80000000
#define SPI_FREQUENCY  27000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  5000000
```

the BL is -1 here,maybe that's one of the cause,and the dimensions:240x240,you're read user showed: 320x240

erikbartmann, 2023-09-08:
Thanks again! I'm really frustrated... It's not working.
Here is the picture of my Arduini-nano-ESP32-Board with the connections:
Pin 11 -> SDA
Pin 10 -> CS
Pin  9 -> DC
Pin 13 -> SCL
[image]

And the content of the setup-file looks like:
`
#define USER_SETUP_INFO "User_Setup"
#define GC9A01_DRIVER

#define TFT_MOSI 11 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 13
#define TFT_CS   10  // Chip select control pin
#define TFT_DC    9  // Data Command control pin
//#define TFT_RST  33  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   -1  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  		27000000
#define SPI_READ_FREQUENCY  5000000
#define SPI_TOUCH_FREQUENCY 2500000
`
I tested it with and without the RST-Signal.
The BL seems to be on.
This is certainly only a small thing...

Thnaks in advance!

Best

Erik

copyCATandpasteDOG, 2023-09-08:
Much better than nothing :)

why don't you try with 
    #define TFT_RST  4,
it could do the trick

erikbartmann, 2023-09-08:
I did it and I the display is still dark :-(

01GOD, 2023-09-08:
> My bad,you're right, i have the same as your's but forgot the pin details,i use it with a esp32 , my setup working:
>
> ```
> //GC9A01 ROUND TFT & ESP32
>
> #define GC9A01_DRIVER
>
>
> #define TFT_MOSI 23 // In [...]

Literally tried that exact code in the User_Setup.h file a few moments ago on an ESP32 DOIT and a display that looks as though almost the same hw. When ran the "animated dial" example, there is blocky graphics tearing. 

Any solution?

copyCATandpasteDOG, 2023-09-08:
Try this clock sketch:

```
/*
 An example analogue clock using a TFT LCD screen to show the time
 use of some of the drawing commands with the ST7735 library.

 For a more accurate clock, it would be better to use the RTClib library.
 But this is just a demo. 

 Uses compile time to set the time so a reset will start with the compile time again
 
 Gilchrist 6/2/2014 1.0
 Updated by Bodmer
 */

#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

#define TFT_GREY 0xBDF7

float sx = 0, sy = 1, mx = 1, my = 0, hx = -1, hy = 0;    // Saved H, M, S x & y multipliers
float sdeg=0, mdeg=0, hdeg=0;
uint16_t osx=130, osy=130, omx=65, omy=65, ohx=130, ohy=130;  // Saved H, M, S x & y coords
uint16_t x0=0, x1=0, yy0=0, yy1=0;
uint32_t targetTime = 0;                    // for next 1 second timeout

static uint8_t conv2d(const char* p) {
  uint8_t v = 0;
  if ('0' <= *p && *p <= '9')
    v = *p - '0';
  return 10 * v + *++p - '0';
}

uint8_t hh=conv2d(__TIME__), mm=conv2d(__TIME__+3), ss=conv2d(__TIME__+6);  // Get H, M, S from compile time

bool initial = 1;

void setup(void) {
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_GREY);
  tft.setTextColor(TFT_GREEN, TFT_GREY);  // Adding a black background colour erases previous text automatically
  
  // Draw clock face
  tft.fillCircle(130, 130, 125, TFT_BLUE);
  tft.fillCircle(130, 120,130, TFT_BLACK);

  // Draw 12 lines
  for(int i = 0; i<360; i+= 30) {
    sx = cos((i-90)*0.0174532925);
    sy = sin((i-90)*0.0174532925);
    x0 = sx*125+130;
    yy0 = sy*125+130;
    x1 = sx*118+130;
    yy1 = sy*118+130;

    tft.drawLine(x0, yy0, x1, yy1, TFT_BLUE);
  }

  // Draw 60 dots
  for(int i = 0; i<360; i+= 6) {
    sx = cos((i-90)*0.0174532925);
    sy = sin((i-90)*0.0174532925);
    x0 = sx*125+130;
    yy0 = sy*125+130;
    
    tft.drawPixel(x0, yy0, TFT_BLUE);
    if(i==0 || i==180) tft.fillCircle(x0, yy0, 2, TFT_CYAN);
    if(i==0 || i==180) tft.fillCircle(x0+1, yy0, 2, TFT_CYAN);
    if(i==90 || i==270) tft.fillCircle(x0, yy0, 2, TFT_CYAN);
    if(i==90 || i==270) tft.fillCircle(x0+1, yy0, 2, TFT_CYAN);
  }

  tft.fillCircle(130, 130, 3, TFT_RED);

  // Draw text at position 64,125 using fonts 4
  // Only font numbers 2,4,6,7 are valid. Font 6 only contains characters [space] 0 1 2 3 4 5 6 7 8 9 : . a p m
  // Font 7 is a 7 segment font and only contains characters [space] 0 1 2 3 4 5 6 7 8 9 : .
  //tft.drawCentreString("Time flies",128,130,4);

  targetTime = millis() + 1000; 
}

void loop() {
  if (targetTime < millis()) {
    targetTime = millis()+1000;
    ss++;              // Advance second
    if (ss==60) {
      ss=0;
      mm++;            // Advance minute
      if(mm>59) {
        mm=0;
        hh++;          // Advance hour
        if (hh>23) {
          hh=0;
        }
      }
    }

    // Pre-compute hand degrees, x & y coords for a fast screen update
    sdeg = ss*6;                  // 0-59 -> 0-354
    mdeg = mm*6+sdeg*0.01666667;  // 0-59 -> 0-360 - includes seconds
    hdeg = hh*30+mdeg*0.0833333;  // 0-11 -> 0-360 - includes minutes and seconds
    hx = cos((hdeg-90)*0.0174532925);    
    hy = sin((hdeg-90)*0.0174532925);
    mx = cos((mdeg-90)*0.0174532925);    
    my = sin((mdeg-90)*0.0174532925);
    sx = cos((sdeg-90)*0.0174532925);    
    sy = sin((sdeg-90)*0.0174532925);

    if (ss==0 || initial) {
      initial = 0;
      // Erase hour and minute hand positions every minute
      tft.drawLine(ohx, ohy, 130, 130, TFT_BLACK);
      ohx = hx*131+130;    
      ohy = hy*131+130;
      tft.drawLine(omx, omy, 129, 129, TFT_BLACK);
      omx = mx*107+130;    
      omy = my*107+130;
    }

      // Redraw new hand positions, hour and minute hands not erased here to avoid flicker
      tft.drawLine(osx, osy, 130, 130, TFT_BLACK);
      tft.drawLine(ohx, ohy, 130, 130, TFT_WHITE);
      tft.drawLine(omx, omy, 130, 130, TFT_WHITE);
      osx = sx*111+130;    
      osy = sy*111+130;
      tft.drawLine(osx, osy, 130, 130, TFT_RED);

    tft.fillCircle(130, 130, 3, TFT_RED);
  }
}
```

Bodmer, 2023-09-08:
The library expects GPIO numbers to be used, not the Dx numbers. Chech the board documents and use the GPIO number relevant to each Dx pin.

The GC9A01 does not support pixel reads.

erikbartmann, 2023-09-09:
Ok, I will sart at the beginning. Here is my circuit.
[image]
I'm using the clock-sketch.
Here are my settings.

#define USER_SETUP_INFO "User_Setup"
#define GC9A01_DRIVER

#define TFT_MOSI 38 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 48
#define TFT_CS   21  // Chip select control pin
#define TFT_DC   18  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   -1  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  		27000000
#define SPI_READ_FREQUENCY  5000000
#define SPI_TOUCH_FREQUENCY 2500000

I'm not sure what to do with the RST-Pin of the TFT-Board.

Best

Erik

copyCATandpasteDOG, 2023-09-09:
Maybe if you connect RST to the reset pin: ->  #define TFT_RST -1

erikbartmann, 2023-09-09:
No, it didn't work.

copyCATandpasteDOG, 2023-09-09:
->I've seen this very interesting game with the modules you have:

https://blog.arduino.cc/2023/08/26/play-full-color-games-on-this-simple-arduino-nano-esp32-based-handheld-console/?queryID=undefined

And on the video the guy shows his own setup:

```
// GC9A01 & ARDUINO NANO ESP32

#define GC9A01_DRIVER

// For ESP32 Dev board (only tested with GC9A01 display)
// The hardware SPI can be mapped to any pins

#define TFT_SCLK 47
#define TFT_MOSI 38 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_RST  21 // Reset pin (could connect to Arduino RESET pin)
#define TFT_DC   18  // Data Command control pin
#define TFT_CS   17 // Chip select control pin
#define TFT_MOSI 10

//#define TFT_BL   22  // LED back-light

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

//#define SPI_FREQUENCY  80000000
#define SPI_FREQUENCY  27000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  5000000
```
Strangely,he use GPIO 47 and not  GPIO 48,and he use TFT_MISO too,and no backlight using;
could be worth trying it.

I tried the "Boing' ball demo" in the DMA repertory ,very impressing with that TFT,why not trying it!

Bodmer, 2023-09-15:
The GPIO pin numbers must be used in the setup file (NOT the Dx references).

To find out which GPIO is on which pin on the board refer to the documents here:

https://docs.arduino.cc/tutorials/nano-esp32/cheat-sheet

Bodmer, 2023-09-25:
The board is demonstrated here:
https://www.youtube.com/watch?v=GjI8mUnlrPM

I suggest you try the same pins as that works for Volos.

Since the board and display combination works for Volos I am unable to advise further.

closed 2023-09-25 by Bodmer