Bodmer/TFT_eSPI #2841: Waveshare ESP32-S3-PICO with Waveshare ResTouch-PICO-LCD-3.5 fails DC/CS out of range

illigtr, 2023-09-09:
Description of the problem.

Waveshare makes a 3.5 LCD with touchscreen called the ResTouch-PICO-LCD-3.5. This HAT supports the 40 pin Pi Pico, Pi Pico W, Waveshare dev kits: ESP32-S2-Pico and WS ESP32-S3-Pico boards. I can successfully use TFT_eSPI with all the combinations except the S3 due to the limitation of the LCD DS and CS requiring a GPIO of 31 or less. I have even tried to breadboard the S3 to the LCD and change DC and CS to a lower GPIO. Still fails. No screen initialization at all. If any user actaully want to use the S3/LCD combination they are forced to use the pin/GPIO listed below.

IDE
    Arduino v2.2.1

TFT_eSPI library version
    v2.5.0

Board package version
   Espresif v2.0.12

Processor
    Waveshare ESP32-S3-PICO

TFT driver
    Waveshare ResTouch-PICO-LCD-3.5 using ILI9488 and XP2066
    
Interface
    SPI

Connections:

Pin    LCD-3.5            PiPico    S2-PICO    S3-PICO     Notes
11     LCD DC           GPIO8     GPIO16     GPIO33     
12     LCD CS           GPIO9     GPIO17     GPIO34     
14     LCD SPI CLK   GPIO10    GPIO10     GPIO35     
15     LCD SPI MOSI  GPIO11    GPIO11     GPIO36     
16     LCD SPI MISO  GPIO12    GPIO12     GPIO37     
17     LCD BL           GPIO13    GPIO13     GPIO38     
20     LCD RST         GPIO15    GPIO35     GPIO40     
21     TP CS              GPIO16    GPIO36      GPIO42     
22     TP IRQ           GPIO17    GPIO37     GPIO41     not used

TFT_eSPI Test          works         works          fails     

For each configuration I have created a custom user setup.h file.

I know that GPIO 33/34 can be used for DC/CS, as I have compiled a Home Assistant sensor with ESPHome using
the WS ESP32-S3-PICO and ResTouch-PICO-LCD-3.5 with success.

illigtr, 2023-09-09:
[image]

Bodmer, 2023-09-09:
These mappings of the SPI pins are supported:
```
                    // Typical board default pins - change to match your board
#define TFT_CS   34 //     10 or 34 (FSPI CS0) 
#define TFT_MOSI 35 //     11 or 35 (FSPI D)
#define TFT_SCLK 36 //     12 or 36 (FSPI CLK)
#define TFT_MISO 37 //     13 or 37 (FSPI Q)

// FSPI port (SPI2) used unless following defined
// #define USE_HSPI_PORT
```
GPIO above 31 can be used for DC, CS and RST

This means the board should work if you swap the SCLK and MOSI connections.