Bodmer/TFT_eSPI #2843: Problem with ESP32 CPU crashing when using library

tdep11, 2023-09-10:
Hello,

I'm using an ILI9341 Touch Screen with the TFT_eSPI library. I have created a code for a kind of scale, with an interface in which you have to select info on the screen. In this project, a pariodic table is drawn, and the user must click on the element he is weighing.

The library works fine, but at some point, inside one specific while loop, when I ask the EPS32 to perform anything on the screen, it crashes.

Here is the error I get:

```
GuruMeditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d4cc9  PS      : 0x00060c30  A0      : 0x800d2a04  A1      : 0x3ffc6820
A2      : 0x3ffc30a8  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x08000000  
A10     : 0x3ffc329c  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000001  
A14     : 0x3ffc30a8  A15     : 0x00000011  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000044  LBEG    : 0x400d9270  LEND    : 0x400d92d4  LCOUNT  : 0x00000000
```

It seems like it is trying to read/write at an illegal adress.
The code is very long, and most of it is irrelevant for my problem, so I put only the essential parts.

```
#include <HX711_ADC.h> 
#include <Wire.h>
#include <EEPROM.h>
#include <molmetre_v2.h>

#include "FS.h"

#include <SPI.h>

#include "Free_Fonts.h"  // Include the header file attached to this sketch

#include <TFT_eSPI.h>  // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

#define LABEL1_FONT &FreeSansBold9pt7b

#define MYGREY 0xDEFB      ///< 224, 224, 224
#define PALE 0xDF9E        ///< 223, 241, 243
#define LIGHTBLUE 0xB6FC   ///< 185, 223, 234
#define MIDBLUE 0x969C     ///< 151, 212, 233
#define DARKBLUE 0x24BB    ///< 36, 148, 224
#define PURPLE 0xB417      ///< 178, 129, 193
#define PINK 0xE294        ///< 231, 80, 162
#define DARKYELLOW 0xF641  ///< 250, 201, 8

#define MOLMETRE "MOLMETRE"
#define MyName "Thibault de Percin"

float MM[57] = {
 //molar masses of all the elements (needed after)
};

char elements[57][3] = {
  //names of all the elements
};

char L[27] = {
  //alphabet in capital letters
};
char l[27] = {
  //alphabet is small letters
};

uint8_t numberElements = 57;

bool completed = 0;
bool nextStep = 0;
bool finished = 0;
uint8_t mode;
float last_i = 0;
int count = 0;
int number = 0;

int formula[5] = {};
int stoechio[5] = {};

// Invoke the TFT_eSPI button class and create all the button objects
TFT_eSPI_Button button[15];
//0: Balance
//1: Molmetre
//2: Tare
//3: Terminé for measure
//4: Rentrer la formule
//5: Espece enregistree
//6: Terminé for periodic table
//7:
//8:

#define HX711_dout 16  //mcu > HX711 dout pin
#define HX711_sck 17   //mcu > HX711 sck pin
HX711_ADC LoadCell(HX711_dout, HX711_sck);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(38400);

//this is for the HX711 module of weight
  LoadCell.begin();
  boolean _tare = true;
  LoadCell.start(2000, _tare);
  LoadCell.setCalFactor(1135.16);  //Find the right value by using the Calibration file in the HX711 examples

  tft.init();
  tft.setRotation(1);

//this calibration data comes from the file Touch_calibrate in the example of the TFT_eSPI library.
  uint16_t calData[5] = { 365, 3546, 239, 3597, 7 };
  tft.setTouch(calData);  //to change this data, reuse the code Touch_calibrate

  tft.fillScreen(ILI9341_BLACK);  //fill screen with black
  yield();                        //I don't know what this does, not sure it's necessary

  tft.setTextDatum(MC_DATUM);
  tft.setFreeFont(&FreeSansBold24pt7b);    //big font (18 or 24)

  // I deleted some lines that are just the home screen

}

void loop() {
  // put your main code here, to run repeatedly:

  uint16_t t_x = 0, t_y = 0;  // To store the touch coordinates
  // Get current touch state and coordinates
  bool pressed = tft.getTouch(&t_x, &t_y);

  if (pressed) {
    doMeasure();
  }
}

void doMeasure() {
  tft.fillScreen(TFT_BLACK);

  nextStep = 0;
  finished = 0;
  count = 0;
  number = 0;
  tft.fillScreen(TFT_BLACK);
  drawTable();
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.drawString("Cliquer sur le premier ", 160, 10);
  tft.drawString("element de la formule", 160, 30);

  for (int i = 0; i < 10; i++) {
    formula[i] = 0;
    stoechio[i] = 0;
  }

  while (finished == 0) {

    nextStep = 0;
    while (nextStep == 0) {
      number = testTable();

      if (count > 2) { //this creates a button that is used to say you have finished choosing the various elements you are weighing
        button[7].initButton(&tft, 140, 80, 110, 40, TFT_BLACK, TFT_WHITE, TFT_RED, "Termine", 1);
        //            x, y (center button), w,    h,    outline,   fill,    text color,    text,   text size multiplier
        button[7].drawButton();

        uint16_t t_x = 0, t_y = 0;  // To store the touch coordinates
        bool pressed = tft.getTouch(&t_x, &t_y);
        if (pressed && button[7].contains(t_x, t_y)) {
          button[7].press(true);  // tell the button it is pressed
        } else {
          button[7].press(false);  // tell the button it is NOT pressed
        }
        if (button[7].justPressed()) {
          nextStep = 1;
          finished = 1 
        }
      }
      Serial.print("Step 2");
    }
    nextStep = 0;

    while (nextStep == 0) {
      Serial.print("Step 6");
      //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
      //                                    Here is the moment it crashes
      //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
      tft.fillRect(0, 0, 1, 1, TFT_BLACK);
      //If I remove this line it doesn't crash
    }

    if (finished == 0) {
      formula[count] = number;
      Serial.print("number:");
      Serial.print(number);
      count++;
    }
  }
}

void drawTable(){
   //this draws the periodic table
}

int testTable() {//This (long) function returns a value between 0 and 56 (the number corresponds to the number of the element chosen)
}

```

As you can see in the code, inside a certain while loop, I put a tft.fillRect(0,0,1,1, BLACK) funtion. If this function is right over (out of the while loop), the code works fine, but whenever I put it into the loop, it crashes immediately (the first time it tries to). In the IDE, I get "Step 6", as asked for in the code, and immediately after, I get the error... I really don't understand why that command makes it crash.
Can anyone help me?

Tell me if there is info that I should remove or that is missing, or if I have done anything wrong!
Thanks!

Bodmer, 2023-09-10:
At the start:
```
int formula[5] = {};
int stoechio[5] = {};
```
Then late:
```
for (int i = 0; i < 10; i++) {
    formula[i] = 0;
    stoechio[i] = 0;
  }
```

Can you see the problem?

tdep11, 2023-09-10:
Yes, I see! I'll correct that and see if it works

tdep11, 2023-09-10:
It works!!!
Thank you so much, that's really a dumb mistake, sorry...

closed 2023-09-10 by tdep11