Bodmer/TFT_eSPI #2848: number ghosting upwards and very poor frame rate with esp32, ili9341, and DMA.

ripjohnbrown1859, 2023-09-13:
```
#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <Wire.h>
#include "MAX30105.h"
#include "spo2_algorithm.h"
#define MAX_GEN_COUNT 500
#define GRIDX 240
#define GRIDY 240
#define CELLXY 1
#define GEN_DELAY 10
uint8_t ignoreBeats = 0;
uint8_t SamplePeriod = 5;
uint8_t grid[GRIDX][GRIDY];
uint8_t newgrid[GRIDX][GRIDY];
uint16_t genCount = 0;
uint16_t readingHistory[240];
uint16_t spO2readingHistory[240];
byte historyPointer;
byte spO2historyPointer;
int historyLength = 240;
TFT_eSPI tft = TFT_eSPI();
// TFT display
MAX30105 spO2Sensor;
unsigned long benchmark = millis();
// Create two sprites for a DMA toggle buffer
TFT_eSprite spr[3] = {TFT_eSprite(&tft), TFT_eSprite(&tft), TFT_eSprite(&tft)};
int bpm = 0;
int spO2 = 0;
int Z = 0;
// Pointers to start of Sprites in RAM (these are then "image" pointers)
uint16_t* sprPtr[3];
unsigned long timer;
uint16_t cursorX = 0;
void calcBPM(uint8_t ecg, uint16_t x)
{
  static uint8_t prevy = 0;
  uint32_t t;
  static uint32_t prevt = 0;
  static int InPeak = 0;
  static int bpm = 50;
  const int threshHi = 40;
  const int threshLo = threshHi - 5;
  int b;
  static bool first = true;

  if (ecg > prevy)
    prevy++; else
    prevy--;

  if (InPeak > 0) {
    InPeak++;
    if (ecg < prevy+threshLo) {     
      t = millis();
      b = 60000 / (t-prevt);
      if ((b >= 20) && (b <= 250) && (InPeak < 100/SamplePeriod)) { // peak must be narrow and BPM must be reasonable
        if (ignoreBeats > 0) {
          ignoreBeats--;
        } else {
          bpm = (bpm*7+b)/8; // smoothing
          bpm += constrain(b-bpm,-1,+1);
          
        }

      }
      prevt = t;      
      InPeak = 0;
    }    
  } else {
    if (ecg > prevy+threshHi) 
      InPeak = 1;
  } 
  spr[2].setCursor(0,10);
  if(bpm<100){
  spr[2].setTextSize(6);
  }
  if(bpm>99){
  spr[2].setTextSize(4);
  }
  spr[2].print(bpm); 
}

void setup() {
  //Set up the display
  tft.init();     // Init ST7735S chip, black tab
  tft.initDMA();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextSize(1);
  tft.setTextColor(ILI9341_WHITE);
  sprPtr[0] = (uint16_t*)spr[0].createSprite(1, 120);
  sprPtr[1] = (uint16_t*)spr[1].createSprite(1, 120);
  sprPtr[2] = (uint16_t*)spr[2].createSprite(120,240);
  spr[0].setTextDatum(MC_DATUM);
  spr[1].setTextDatum(MC_DATUM);
  spr[2].setTextDatum(TL_DATUM);
  spO2Sensor.begin(Wire, I2C_SPEED_FAST);
    //Setup to sense a nice looking saw tooth on the plotter
  byte ledBrightness = 0x1F; //Options: 0=Off to 255=50mA
  byte sampleAverage = 8; //Options: 1, 2, 4, 8, 16, 32
  byte ledMode = 3; //Options: 1 = Red only, 2 = Red + IR, 3 = Red + IR + Green
  int sampleRate = 100; //Options: 50, 100, 200, 400, 800, 1000, 1600, 3200
  int pulseWidth = 411; //Options: 69, 118, 215, 411
  int adcRange = 4096; //Options: 2048, 4096, 8192, 16384
  uint32_t irBuffer[100]; //infrared LED sensor data
  uint32_t redBuffer[100];  //red LED sensor data 
  spO2Sensor.setup(ledBrightness, sampleAverage, ledMode, sampleRate, pulseWidth, adcRange); //Configure sensor with these settings
  pinMode(26, INPUT);
  Serial.begin(9600);
  tft.startWrite();
}

void loop() {
  if(cursorX > 240){
    cursorX = 0;
  }
  spr[0].drawFastVLine(0, 0, 120, ILI9341_BLACK);
  spr[1].drawFastVLine(0, 0, 120, ILI9341_BLACK);
  //spr[2].fillSprite(ILI9341_BLACK);
  int acquiredValue = analogRead(26);
  int sensorValue = map(acquiredValue, 0, 4025, 0, 120);
  int spO2sensorValue = map(spO2Sensor.getRed(), 115000, 120000, 0, 120);
  Z = spO2Sensor.getRed()/spO2Sensor.getIR();
  spO2 = ((-45.06*Z + 30.354)*Z + 94.845);
  spr[2].setTextSize(4);
  spr[2].drawNumber(spO2, 0, 140);
  spr[0].drawPixel(0, sensorValue, ILI9341_RED);
  spr[1].drawPixel(0, spO2sensorValue, ILI9341_BLUE);
  spr[2].setTextSize(1);
  spr[2].drawNumber((millis() - benchmark), 0, 0);
  benchmark = millis();
  calcBPM(acquiredValue, 0);
  tft.pushImageDMA(cursorX, 0, 1, 120, sprPtr[0]);
  tft.pushImageDMA(cursorX, 120, 1, 120, sprPtr[1]);
  tft.pushImageDMA(240, 0, 120, 240, sprPtr[2]);
  cursorX = cursorX + 1;
}
```
I get ghosted versions of the numbers appearing above the designated positions on the right hand side of the screen, also whenever I try to fill the sprite with black it only displays the only the top half of the characters drawn at 0,0 for in the top left hand of the sprite. I am running an ILI9341 and nodemcu 32s in platform IO using SPI at 40mhz. library version 2.5.31,

Bodmer, 2023-09-13:
The sketch should loop at around 80fps (12ms per loop).

I suspect acquiring the sensor reading is slowing the loop down. Comment out part of those lines as a test, eg:
  int spO2sensorValue = 123;  //map(spO2Sensor.getRed(), 115000, 120000, 0, 120);
  Z = 321; //spO2Sensor.getRed()/spO2Sensor.getIR();

The sprite is still being drawn by the DMA engine when it is being cleared, so that will lead to corruption of the display. Wait for DMA to finish updating the screen before clearing the sprite:
  tft.dmaWait();
  spr[2].fillSprite(ILI9341_BLACK);

You could acquire sensor data while DMA is in progress, so move the above 2 lines after the sensor read lines.

ripjohnbrown1859, 2023-09-14:
thanks! i moved the fill and sped up the rate on the sensor as much as i could and i got it up to 62.5 fps with no ghosting.

closed 2023-09-14 by Bodmer