Bodmer/TFT_eSPI #2858: Super bare-bones example uses too much memory

SpamWall, 2023-09-21:
When compiling the following super-basic code I get this error message:

FQBN: arduino:avr:nano:cpu=atmega328old
Using board 'nano' from platform in folder: C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...
C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\sketch\ammeter.ino.cpp -o nul
Alternatives for TFT_eSPI.h: [TFT_eSPI@2.5.0]
ResolveLibrary(TFT_eSPI.h)
  -> candidates: [TFT_eSPI@2.5.0]
C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs -IC:\Users\spamw\OneDrive\Documents\Arduino\libraries\TFT_eSPI C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\sketch\ammeter.ino.cpp -o nul
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs -IC:\Users\spamw\OneDrive\Documents\Arduino\libraries\TFT_eSPI -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\sketch\ammeter.ino.cpp -o nul
Using cached library dependencies for file: C:\Users\spamw\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp
Using cached library dependencies for file: C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src\SPI.cpp
Generating function prototypes...
C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\variants\eightanaloginputs -IC:\Users\spamw\OneDrive\Documents\Arduino\libraries\TFT_eSPI -IC:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI\src C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\sketch\ammeter.ino.cpp -o C:\Users\spamw\AppData\Local\Temp\3595982065\sketch_merged.cpp
C:\Users\spamw\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\spamw\AppData\Local\Temp\3595982065\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "-IC:\\Users\\spamw\\OneDrive\\Documents\\Arduino\\libraries\\TFT_eSPI" "-IC:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SPI\\src" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7\\sketch\\ammeter.ino.cpp" -o "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7\\sketch\\ammeter.ino.cpp.o"
Compiling libraries...
Compiling library "TFT_eSPI"
Using previously compiled file: C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\libraries\TFT_eSPI\TFT_eSPI.cpp.o
Compiling library "SPI"
Using previously compiled file: C:\Users\spamw\AppData\Local\Temp\arduino\sketches\C1929A710D45A5F1837BC68BCF4F67A7\libraries\SPI\SPI.cpp.o
Compiling core...
Using precompiled core: C:\Users\spamw\AppData\Local\Temp\arduino\cores\arduino_avr_nano_cpu_atmega328old_dd62aba6fe58242e6ba1f785e4240bda\core.a
Linking everything together...
"C:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.elf" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7\\sketch\\ammeter.ino.cpp.o" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7\\libraries\\TFT_eSPI\\TFT_eSPI.cpp.o" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7\\libraries\\SPI\\SPI.cpp.o" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/..\\..\\cores\\arduino_avr_nano_cpu_atmega328old_dd62aba6fe58242e6ba1f785e4240bda\\core.a" "-LC:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7" -lm
"C:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.elf" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.eep"
"C:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.elf" "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.hex"

Using library TFT_eSPI at version 2.5.0 in folder: C:\Users\spamw\OneDrive\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 1.0 in folder: C:\Users\spamw\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\SPI 
"C:\\Users\\spamw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\spamw\\AppData\\Local\\Temp\\arduino\\sketches\\C1929A710D45A5F1837BC68BCF4F67A7/ammeter.ino.elf"
Sketch uses 35392 bytes (115%) of program storage space. Maximum is 30720 bytes.
Global variables use 425 bytes (20%) of dynamic memory, leaving 1623 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board
--------------------------------------

Arduino IDE V2.2.1
Arduino NANO

TFT_eSPI library version 2.5.0
Waveshare 1.28" round display LCD screen module 240×240 pixels SPI 65K color
GC9A01A
SPI

```
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(9600);

  tft.init();
  tft.fillScreen(TFT_BLUE);
}

void loop(void) {
  delay(100);
}
```

Pablo2048, 2023-09-22:
From README.md `A feature rich Arduino IDE compatible graphics and fonts library for 32 bit processors.` Atmega328 is 8 bit CPU so...

Bodmer, 2023-09-22:
The library is intended for higher performance processors with lots more FLASH and RAM.

It can run on an 8 bit processor but you will need to comment out the extra fonts listed in the setup file.

closed 2023-09-22 by Bodmer