Bodmer/TFT_eSPI #2860: Problem with ST7789V3 172*320 display

mason334, 2023-09-24:
Hi there, I cannot get my 1.47 inch ST7789V3 172*320 display to work. 

Here is my setup:
#define ST7789_DRIVER

#define TFT_DC   0  // Data Command control pin
#define TFT_RST  1  // Reset pin (could connect to RST pin)
#define TFT_MOSI 2  // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 3
#define TFT_CS   4  // Chip select control pin

With the same setup, I can get a 1.3 inch 240*240 ST7789 display to work with no problem. 

The only difference is ST7789 does not have a CS pin, where ST7789V3 has a CS pin. 

Thanks for any suggestions and comments.

Bodmer, 2023-09-24:
Run Read_User_Setup and look at the serial output to check the compiler is picking up the correct settings.

The display will have an offset so you need to specify the display size in the setup file:

#define TFT_WIDTH  170
#define TFT_HEIGHT 320

Is the backlight on?

mason334, 2023-09-25:
Thanks
Problem solved
Library can support st7789v3 with no problem
[https://github.com/mason334/images/blob/main/st7789.jpg](url)

closed 2023-09-25 by Bodmer