Bodmer/TFT_eSPI #2870: ESP32 SPI share error : TFT ST7735 & MAX31865

pploume, 2023-10-01:
Hi,

I cant have working together 2 SPI devices on the same bus (esp wroom 32) :
- TFT ST7735
- Max31865 (from adafruit like board) for temperature reading

Individually, the 2 devices are working OK
They share MOSI/MISO/CLK, default from user_Setup.h
The 2 CS are different (tempCS --> 13, tftCS --> 12)

If I init the TFT first, the display works until the max31865 inits. Then, no more tft
If I init the Max31865 first, the temperature is OK until the TFT inits. Then no more temperature

I saw the issue "Add SPIClass object parameter for <init> function #1991" ... but links in the posts are not reachable (and not found the project on author's page).
And not sure I understand how and when call the spi_start() + spi_end() proposed function

I'm using the #include <adafruit_max31865.h> library ...
I saw on github some others librairies ... not tryied 

Any help appreciated, and please consider I'm a noob
:)

pploume, 2023-10-01:
Solved by constructing the max31865 with only CS (and not other SPI ...)

hope it helps other guys

Bodmer, 2023-10-01:
Glad you have solved it, although I don't understand what you did to fix the compatibility issue!

The problem occurs because each library is using a different SPI instance attached to the same hardware port.

closed 2023-10-01 by Bodmer

pploume, 2023-10-02:
Hi,

I didn't dig that much to understand ...

I think that initializing the TFT with your library creates an instance of SPI
this instance being re-used by the adafruit's one when calling its constructor without SPI pins (and only CS) 

In this way, and as you mentionned it, there is only one SPI used