Bodmer/TFT_eSPI #2884: Been getting a GPIO compiling error code after the latest ESP-IDF update

mikedalli, 2023-10-08:
Been getting this compiling error code in the Arduino IDE after the recent ESP-IDF update. I am using TFT_eSPI on a Lilygo T-Display S3. Tried all the fixes I found online but nothing worked. I might be doing something wrong as I just started programing for the last 6 months. Is there anything I can do to fix this issue?

In file included from c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:96,
                 from c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:16:
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c: In function 'void dc_callback(spi_transaction_t*)':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:203:22: error: 'GPIO' was not declared in this scope
  203 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c:783:28: note: in expansion of macro 'DC_D'
  783 |   if ((bool)spi_tx->user) {DC_D;}
      |                            ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.c:784:9: note: in expansion of macro 'DC_C'
  784 |   else {DC_C;}
      |         ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:247:22: error: 'GPIO' was not declared in this scope
  247 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:80:5: note: in expansion of macro 'CS_L'
   80 |     CS_L;
      |     ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::begin_nin_write()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:247:22: error: 'GPIO' was not declared in this scope
  247 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:92:5: note: in expansion of macro 'CS_L'
   92 |     CS_L;
      |     ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:248:22: error: 'GPIO' was not declared in this scope
  248 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:106:7: note: in expansion of macro 'CS_H'
  106 |       CS_H;
      |       ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::end_nin_write()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:248:22: error: 'GPIO' was not declared in this scope
  248 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:121:7: note: in expansion of macro 'CS_H'
  121 |       CS_H;
      |       ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:247:22: error: 'GPIO' was not declared in this scope
  247 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:141:5: note: in expansion of macro 'CS_L'
  141 |     CS_L;
      |     ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:248:22: error: 'GPIO' was not declared in this scope
  248 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:161:7: note: in expansion of macro 'CS_H'
  161 |       CS_H;
      |       ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::writecommand(uint8_t)':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:964:3: note: in expansion of macro 'DC_C'
  964 |   DC_C;
      |   ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:203:22: error: 'GPIO' was not declared in this scope
  203 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
[... 36 lines cut ...]
 3495 |   DC_C; tft_Write_8(TFT_CASET);
      |   ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::drawPixel(int32_t, int32_t, uint32_t)':
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3677:7: note: in expansion of macro 'DC_C'
 3677 |       DC_C; tft_Write_8(TFT_CASET);
      |       ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3684:7: note: in expansion of macro 'DC_C'
 3684 |       DC_C; tft_Write_8(TFT_PASET);
      |       ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32_S3.h:202:22: error: 'GPIO' was not declared in this scope
  202 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
c:\Users\mdall\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:3690:3: note: in expansion of macro 'DC_C'
 3690 |   DC_C; tft_Write_8(TFT_RAMWR);
      |   ^~~~

exit status 1

Compilation error: exit status 1

mikedalli, 2023-10-08:
I seemed to have missed this and tried it out,  https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L14

It fixed most of the problems up to this error code

c:/users/mdall/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\mdall\AppData\Local\Temp\arduino\sketches\841A8F0E7A697322F0A52D8D28E08E05\libraries\FS\vfs_api.cpp.o:(.literal._ZN11VFSFileImplD2Ev+0x4): undefined reference to `vtable for fs::FileImpl'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Bodmer, 2023-10-11:
The error seems to be with the FS library:

...\libraries\FS\vfs_api.cpp...

It might be something in your sketch is the root cause.

The latest master library should not generate the other errors. Which version of TFT_eSPI are you using?

mikedalli, 2023-10-11:
Thank you very much for the response and also for a great library. The library version I am using is 2.5.0.  As for the code its the very same code I use for the begining of every project I use for a lillygo t_display s3. Below is the code quoted. I tried it in both ways unquoted and quoted both with the same result. I even tried it with the quoted part deleted just to make sure its not reading the quoted part (I know I am green but I had to give it a shot) . I tried the button read sketch on its own and everything works but as soon as I "#include <TFT_eSPI.h>" the eror pops up.

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

//#define PB1 0
//#define PB2 14
//bool button1 = 1;
//bool button2 = 1;

//int brightness = 255;

void setup() {

  /////////////////////
  //Battery Settings///
  //pinMode(15,OUTPUT);
  //digitalWrite(15,1);
  /////////////////////
  tft.init();
  tft.fillScreen(TFT_BLACK);
  //tft.drawString(String(button1), 5, 300, 4);
  //tft.drawString(String(button2), 155, 300, 4);

  //ledcSetup(0, 10000, 8);
  //ledcAttachPin(38, 0);
  //ledcWrite(0, brightness);

  //tft.fillRectHGradient(int16_t x, int16_t y, int16_t w, int16_t h, uint32_t color1, uint32_t color2)
}

void loop(){

  //ButtonRead();
}

/*void ButtonRead(){

  static bool lastState1 = 1;
  static bool lastState2 = 1;
  static unsigned long buttonTimer = 0;
  static int buttonDelay = 1;

  if(millis() - buttonTimer > buttonDelay){
    buttonTimer = millis();
    button1 = digitalRead(PB1);
    button2 = digitalRead(PB2);
    if(button1 != lastState1){
      lastState1 = button1;
      if(lastState1 == 0){
        tft.drawString(String(button1), 5, 300, 4);
      }else{tft.drawString(String(button1), 5, 300, 4);}
    }
    if(button2 != lastState2){
      lastState2 = button2;
      if(lastState2 == 0){
        tft.drawString(String(button2), 155, 300, 4);
      }else{tft.drawString(String(button2), 155, 300, 4);}
    }
  }
}*/

Bodmer, 2023-10-11:
I have created a new release 2.5.31, try that.  It may be 24hrs before te Arduino IDE picks up the new release in the the library manager.

mikedalli, 2023-10-12:
Everything is the same as in my first 2 posts  with release 2.5.31. I even had to add  #include "hal/gpio_ll.h" in the processor sketch, because it gave me the first error code and after that it gave me the 2nd error code.

pablobatch, 2023-10-16:
This is happening to me too with my Lilygo T-Display, since I updated the libraries last friday. 
I also tried with the the 2.5.31 version but the problem remains.
Is there any other workaround to solve this ?

ChrisZang0815, 2023-10-17:
I have the same problem (also on 2.5.31)
Board -> ESP32 Dev Module
And here is my "sketch" which fails:

#include <TFT_eSPI.h>

void setup() {
}

void loop() {
 }

ChrisZang0815, 2023-10-17:
i read in another thread that someone fixed it by using this boards manager:
https://dl.espressif.com/dl/package_esp32_index.json
I did that too, deleted the esp32 directory from "C:\Users\chris\AppData\Local\Arduino15\packages"
deleted the old reference and added the one above 
Restarted Arduino and allowed the package to install - no dice same error

ChrisZang0815, 2023-10-17:
OK, I did a bit more of my own research and first downgraded to 2.5.0 and then 2.4.50
In both cases I get the exact same errors.
I don't think that there's an error in the library, but in some environment that library includes

ChrisZang0815, 2023-10-17:
OK, final (?) update: I reloaded the esp32 boards again and this time it worked

So here is what I did:
1) open Arduino IDE
2) delete the line "https://dl.espressif.com/dl/package_esp32_index.json": from File -> Preferences -> additional board URL
3) close IDE
4) delete the directory esp32 in: "C:\Users\chris\AppData\Local\Arduino15\packages"
5) start IDE
6) File -> Preferences -> additional board URL, add "https://dl.espressif.com/dl/package_esp32_index.json"
it now looks like this:
**https://dl.espressif.com/dl/package_esp32_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json
https://raw.githubusercontent.com/MHEtLive/arduino-boards-index/master/package_mhetlive_index.json**
7) close IDE
8) start IDE
9) Tools -> Boards Manager
10) search for "esp32"
11) Install "esp32 by espressif systems"
12) voila

hope that helps.

mikedalli, 2023-10-17:
Basically you are just running version 2.0.14 on the esp-idf. It compiles but when I try to print something to the display it doesn't show. The problem is with version 3.0.0 alpha2. First there is something missing with the GPIOs and when you manually include that file in the processor file it gives me another error code. Have you tried printing something to the display with version 2.0.14?

lewisxhe, 2023-10-17:
espressif Arduino version 3.0 has made major changes. Please roll back the version to below 3.0. It works well in 2.0.14.

mikedalli, 2023-10-18:
> espressif Arduino version 3.0 has made major changes. Please roll back the version to below 3.0. It works well in 2.0.14.

I did all that and as I stated in previous post. it doesnt print to screen. Only thing I managed to control is the backlight brightness. I found that out after seeing that the screen did not light up after flashing it so I figured let me see if the brightness was the problem.  Made a simple code to fill screen white and nothing shows on the display. Seems to me its not reading the GPIO that connect to the screen. I might be doing something wrong also as I am newbie to all of this. This is the sketch I have to test if it works.

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

int brightness = 255;

void setup() {

ledcSetup(0, 10000, 8);
ledcAttachPin(38, 0);
ledcWrite(0, brightness);

tft.init();
tft.fillScreen(TFT_WHITE);
}
void loop(){
}

Bodmer, 2023-10-18:
Run the Read_User_Setup example and check the serial monitor output is correct.

JohanBenko, 2023-10-19:
I (only) deleted the folder ESP32 as suggested by Chris and after restarting the Arduino IDE 2.1.0. I've been prompted to either manually or automatically update the ESP32 library. While choosing "Automatically" the version 2.0.11 got installed. It compiles without errors!
The problems did arise after upgrading to Arduino IDE 2.2.1 this morning. Now it automatically reverted to version 2.1.0 again (?)

Bodmer, 2023-10-21:
OK, thanks for the update.   I think the latest Arduino IDE loaded ESP32 board package is 2.0.14 and that works fine for me with IDE 2.2.1 and the older 1.8.x

closed 2023-10-23 by Bodmer