Bodmer/TFT_eSPI #2890: ILI9341 Driver Peculiarities

CHCHmark, 2023-10-11:
Hi All, I am a newbie. I hav4 4 TFT screens from china that purport to be 240x320 and driven by the ILI9341 chip. I am working  with an ESP32 WRRoom on Platform IO and Arduino and I have had many other screen working perfectly with the wonderful driver code.
 TFT_eSPI library version  bodmer/TFT_eSPI@^2.4.31
 TFT driver Setup_ILI9341_
SPI 
 TFT to processor connections used TFT_CS= 5  TFT_DC= 25

[image]

This photo shows my issue. It appears the screen dimensions are not quite correct and the colours are Red where they should be Blue and vice versa.
The display appears to be working in all ways so I am hoping this is a coding tweak as opposed to a faulty display.

Any thought gladly received.

Cheers

Mark

Bodmer, 2023-10-12:
Run the Read_User_Setup sketch and post the serial monitor output.

The displays don't appear to be ILI9341 types. Can you get example code from the vendor?

Post a link to the display sales page.

CHCHmark, 2023-10-12:
[code]
TFT_eSPI ver = 2.5.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 5
TFT_DC   = GPIO 25
TOUCH_CS = GPIO 33

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

https://ae04.alicdn.com/kf/H616e8d76d0fa407cb3256ec2f2c81514l.jpg_640x640.jpg

Sorry no examples available

Bodmer, 2023-10-13:
That all looks good. One possibility is that it is a landscpe format display.

Try:

#define ILI9342_DRIVER

copyCATandpasteDOG, 2023-10-14:
You can also test with ILI9486,ILI9488 and ST7796,i noticed some similarities between them,as i can make
them all get working using the ILI9486 setup,but colors and dimensions change :)

CHCHmark, 2023-10-14:
Hi Bodmer
Yes you were correct. A change to #define

Mark Weir

CHCHmark, 2023-10-14:
Hi Bodmer,  Many thanks for your help. I started a message earlier and messed it up. The change of #define driver cured my problem with the screen layout. I had to set the rotation to 0 also. I works well. I now need to change the colour order as the Red and Blue are transposed. Is this some thing  I can change? I have done this successfully before with the ST7735 driver.  I also want to thank you for this excellent library, the support and the very useful examples. Your effort has changed what looks quite scary into a great experience for experimenters.

Bodmer, 2023-10-15:
Try one of these lines in the setup.  I am not sure if it is enabled to work with ILI9341/2 displays though:

#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
or:
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

CHCHmark, 2023-10-15:
Hi Bodmer

Sorry no joy with that.
I can work with it for now

Thanks again for your help

Mark Weir

Bodmer, 2023-10-15:
It works for me, so try again of manually change the setting to TFT_MAD_RGB here:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Defines.h#L66

CHCHmark, 2023-10-15:
Hi Bodmer,
 Once again you are so right. I foolishly tried to put the #define in my sketch setup not really thinking.

Thanks again

Mark

closed 2023-10-16 by Bodmer