Bodmer/TFT_eSPI #2893: Touch convertRawXY() returns wrong result for X or Y less than touchCal value

palmerr23, 2023-10-13:
Using the Touch_calibrate example with an ILI9488 and XPT2016 controller.

touch_calibrate() on my screen returns the following, when the calibration points have been touched **SLIGHTLY INSIDE** the screen border.

`uint16_t calData[5] = { 269, 3508, 285, 3556, 4 };`

If, say,` getTouchRaw(&x, &y)` returns x == 268, 
then `convertRawXY(&x, &y) ` will return x == 65534 
as the calculation `(x_tmp - touchCalibration_x0)` in line 202 of Touch.cpp returns a negative value.
Similarly for y axis touches.

At the other end of the calibration range, convertRawXY( ) returns a value slightly outside the screen dimensions, which is usually not problematic.

Simple fixes would be to set the value of `(x_tmp - touchCalibration_x0)` to 0 if it's negative,
OR to constrain `(x_tmp - touchCalibration_x0)`, and its y equivalent, to a range of [0..4096].

BTW, thanks for such a useful library!

Bodmer, 2023-10-13:
The returned result is deliberately made out of range.

You can use convertRawXY() in a sketch but the result must be checked in the sketch. Changes to the function would break the valid touch check here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.cpp#L188

So, no changes to the library are required assuming I have understood the problem correctly.

palmerr23, 2023-10-13:
Bodmer,

You're quite correct. I forgot to also test for a valid result from getTouch() after using validTouch() as my initial test.

As you say, no changes are required. Please close the issue.

Thanks again for this, and your other, libraries.

closed 2023-10-13 by palmerr23