Bodmer/TFT_eSPI #2894: Sprites and Fonts from SD card

Thecactusman0, 2023-10-13:
When using custom fonts from an SD card with sprites my ESP32-S3 crashes and restarts. I am using Arduino IDE with version 2.5.31 of this library and 2.0.13 of the ESP32 boards. Im using a st7789 display (https://www.buydisplay.com/2-inch-240x320-ips-tft-lcd-display-module-for-arduino-and-raspberry-pi) on the same SPI bus as the sd card, SPI is at 80mhz. Here is the pinout:

TFT_MOSI 36 
TFT_SCLK 37
TFT_CS   38
TFT_DC   39  
TFT_RST  35  
TFT_BL   40  
MISO 36
SD_CS 2

This is the code im using, its just the sd smooth fonts example but modified for my pins and using a sprite. Testing without a sprite works flawlessy but when I introduce the sprite it all falls apart.

```
// Font file is stored on SD card
#include <SD.h>

// Graphics and font library
#include <TFT_eSPI.h>
#include <SPI.h>
#define misoPin 42       // Custom MISO pin
#define mosiPin 36       // Custom MOSI pin
#define sckPin 37        // Custom SCK pin
TFT_eSPI tft = TFT_eSPI();  // Invoke library
TFT_eSprite fb = TFT_eSprite(&tft);   // Sprite object unselected text

// -------------------------------------------------------------------------
// Setup
// -------------------------------------------------------------------------
void setup() {
 
SPI.begin(sckPin, misoPin, mosiPin);
  // Initialise the SD library before the TFT so the chip select is defined
  SD.begin(2);

  // Initialise the TFT after the SD card!
  tft.init();
  tft.setRotation(0);
  tft.fillScreen(TFT_BLACK);
  fb.createSprite(240, 320);
  fb.fillSprite(TFT_BLACK);
  fb.setTextWrap(false);

}

// -------------------------------------------------------------------------
// Main loop
// -------------------------------------------------------------------------
void loop() {

  // Font and background colour, background colour is used for anti-alias blending
  fb.setTextColor(TFT_WHITE, TFT_BLACK);

  fb.loadFont("NotoSansJP-VariableFont_wght15", SD);       // Use font stored on SD

  // Display all characters of the font
  fb.setCursor(10,10);
  fb.print("Beans");
  fb.pushSprite(0,0);
 
}
```
Any help would be appreciated

Bodmer, 2023-10-13:
I suspect this is a memory resource problem. How many characters are in the font? Is it a large file?

Does it work with a smaller sprite?

Thecactusman0, 2023-10-23:
The font file on the sd card is 3.2MB and it contained a couple hundred characters. It was a cut down version of Japanese characters. Initialising the sprite as 100 wide and 40 tall stopped the restart/crash issue somewhat but still doesn't display the text.  I can get some shapes drawn after the pushsprite by using the tft object directly but they have weird artefacts with a band cut out of them.

Bodmer, 2023-10-23:
I think there will be a conflict, since the SD card and TFT are on the same SPI bus and using different SPI instances, thus SPI bus access semaphores fail to manage the hardware access.

Delete these 3 lines in setup():
```
SPI.begin(sckPin, misoPin, mosiPin);
SD.begin(2);
tft.init();
```

Replace with these 4 lines in this exact order:
```
pinMode(2, OUTPUT);
digitalWrite( 2, HIGH);    // Ensure SD card chips select is disabled
tft.init();
SD.begin( 2, tft.getSPIinstance() );
```

Hopefully that will ensure both SD and the TFT libraries use the same SPI instance and play nice together.

Thecactusman0, 2023-10-24:
I agree with your theory as bus conflicts have been a problem, however replacing the code unfortunately only drew the default font. I also defined the miso pin in the user setup header file but that just lead to nothing being drawn at all. A quick test I ran showed it had however initialised the sd card.

Bodmer, 2023-11-06:
It looks like there is an incompatibility problem somewhere.

Can you create a complete example including any required files so that I can test. Post this as a zip file.

If the font file fits into FLASH then that would be a good option and will enhance rendering performance.

Thecactusman0, 2023-11-07:
[Files.zip](https://github.com/Bodmer/TFT_eSPI/files/13277091/Files.zip)

I've attached the code I've been messing with to test this, along with my user setup, the processed font file and original font file. I have found that the SD card is being initialised with the get spi instance method but as soon as the font is set it crashes. With the get spi instance initialisation it also now crashes when trying to draw the font directly to the screen without a sprite. Unfortunately the font file is way too big to be able to fit into the flash

Bodmer, 2023-11-11:
I have thought about this some more and concluded that it will not be possible to use fonts on an SD card if both TFT and SD card are on the same bus. This is because the library tries to keep both the display SPI channel open and the SD card channel to gain access to the glyph data.

The file included above has both vlw and ttf file. If you are using the smooth font capability of the library then the ttf file is not needed and the vlw file will easily fit in FLASH.  Other than that you could try with the SD card and TFT on different SPI busses but this is not something I have tried recently.

closed 2023-11-11 by Bodmer