Bodmer/TFT_eSPI #2900: TFT_espi missing a close bracket on the 10/17/2023 update 2.5.31

MikeyMoMo, 2023-10-16:
This cost me 3 hours of looking at my code to see what I had done.  Turns out, I had not done anything wrong.  I was getting aborts in the executable code and compile errors, too.  What a mess.

Then I realized that updates went on.  When I back-level to version=2.5.0 of tft_espi "my" problem went away.  I am not using sprites or anything exotic.  This normally great library has a problem that does not show up in the errors list.  It shows me I have an error with every one of my routines in my code.  So, I cannot offer any hint where the problem is.

pablobatch, 2023-10-16:
Since the update nothing what I created for my Lilygo T-Display is working. 
I tried to come back to an older version of the library but the problem remains.

Bodmer, 2023-10-16:
Don't forget that your setup selection will be over-written when there is a local library change.

Run Read_User_Setup to check setting the compiler uses.

Bodmer, 2023-10-16:
I have installed a fresh copy of the TFT_eSPI library via the Arduino IDE library manager, selected the [T-Display setup for an ESP32 board](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L58).  Compiled and uploaded a test sketch and it is working fine.

I need more information on the software and hardware environment that is showing this problem. (Processor, IDE version, board package version etc.

pablobatch, 2023-10-17:
After installing ESP32 and TFT driver, it finally worked.
Anyway, I selected this option on the User_Setup_Select.h: 
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT

Bodmer, 2023-10-17:
Great. I don't have the S3 variant of the T display so it is good to know it works and no updates are required in the library.

closed 2023-10-17 by Bodmer