Bodmer/TFT_eSPI #2901: Mutli Display touch, CS problem

SaKiEQ, 2023-10-17:
IDE: Arduino
MCU: ESP32
Display: 320x240 2.8" touch display with ILI9163_DRIVER

I am trying to run 3 identical displays, where each is supposed to touch capability.  Essentially its 3x button arrays, so no complex touch reads are needed. 

How would one go about to select each touch_CS individually ? I do understand the concept for each displays - in sketch CS control - but the same does not seem to work the same way for touch... 

I read in an other issue, that defining the CS_touch with -1 would be one part to the puzzle, is that so CS is not enabled by library and 
can be set manually per display ?

Bodmer, 2023-10-17:
The library does not support multiple touch panels.

There are other libraries that support touch, so you maybe you could use one of those and create an instance for each display?

This is not something I can help you with.

closed 2023-10-17 by Bodmer

SaKiEQ, 2023-10-17:
Thanks. 

Never  mind, I just rewrote the lib a bit the brute way - it now has definitions and functions for 3 TOUCH_CS. When I get the time I might improve it a bit by making it more dynamic. I might even give the main TFT functions a stab, but for now it selective TFT_CS control works I would stick with how it is.