Bodmer/TFT_eSPI #2926: Help needed for clock example going backward, and garbage in background

N0ury, 2023-10-26:
Hi, this is not an issue, it's more a newbie question.
I'm using a X-NUCLEO-GFX01M2 STM display board along with a STM NUCLEO-F411RE board.
Display is a ST7789V (2.2" SPI QVGA TFT LCD 320x240).
It has also a 64-Mbit SPI NOR Flash memory for storing graphic images, texts and texture (I don't use it for the moment).
I use these boards under platformIO.
I try to use the clock example.
Clock is displayed, but seconds needle goes backward (ie from right to left).
Furthermore, the background is not clean on the right side.
I attach a screenshot.
[image]

The clock is not centered, not sure if this is a problem.
It's my very first test of this lib, and I'm very happy to see the clock.
But I would like to go further and deeper in testing and using the lib.
I would also like why this clock wants me to become younger...
And also why the background is not clean.
Here's my plaformio.ini file, it's very simple:
```
[env]
platform = ststm32
board = nucleo_f411re
framework = arduino
monitor_port = hwgrep://0483:374b
upload_port = hwgrep://0483:374b
monitor_speed = 115200
lib_deps = TFT_eSPI
	SPI
build_flags =
  -D STM32=1
  -D NUCLEO_64_TFT=1
  -D STM_PORTA_DATA_BUS=1
  -D STM_PORTB_DATA_BUS=1
  -D ST7789_DRIVER=1
  -D USER_SETUP_LOADED=1
  -D ILI9163_DRIVER=1                           ; Select ILI9163 driver
  -D TFT_WIDTH=320                              ; Set TFT size
  -D TFT_HEIGHT=240
  -D TFT_MISO=PA6                               ; Define SPI pins
  -D TFT_MOSI=PA7
  -D TFT_SCLK=PA5
  -D TFT_CS=PA9
  -D TFT_DC=PB10                                ; Data/Comand pin
  -D TFT_RST=PA1                                ; Reset pin
  -D SPI_FREQUENCY=27000000                     ; Set SPI frequency

[env:main]
    build_src_filter = +<main.cpp>

[env:ex]
    build_src_filter = +<ex.cpp>
```

I'm not sure what to put as SPI_FREQUENCY, I leaved it unchanged. Is this the problem?

This lib is very promising for me. I really want to check it more.

Thanks for any help.

N0ury, 2023-10-26:
Reading back my post I realized there was an error, there was a useless line `-D ILI9163_DRIVER=1 `
I have removed it.
The result is not better.
Here it is in this screenshot:
[image]
Seconds needle has disappeared, and garbage is still here. Not at the same place.
Colors have changed too.

N0ury, 2023-10-26:
Sorry for all this noise.
Looking very closely, now seconds needle goes the right way. It's only in very light yellow on a white background, so near invisible.
But garbage is still here.
Half of the problem is solved...

N0ury, 2023-10-26:
doc says: // For ST7789, ST7735, ILI9163 and GC9A01 ONLY, define the pixel width and height in portrait orientation
so changed to
  -D TFT_WIDTH=240                              ; Set TFT size
  -D TFT_HEIGHT=320
Garbage has gone away...

Clock is not centered vertically and colors are bad.
Is it wanted or is it an issue. I can't say.

Bodmer, 2023-10-26:
By default the sketch centres the clock in portrait mode, not landscape. The setRotation function defines the screen rotation.

The colours are inverted, (black = white).  Run the included example diagnostic sketch [Colour_Test](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Test%20and%20diagnostics/Colour_Test/Colour_Test.ino), see header and use  get black and white correct.

You will then have to use ***one*** of these lines in your sketch after tft.init():

tft.invertDisplay( true );
tft.invertDisplay( false );

Your display also has red and green swapped, so include ***one*** of these lines in your setup, compile, upload and try the Colour_Test example again to find the correct option:

-D TFT_RGB_ORDER TFT_RGB=1 ; Colour order Red-Green-Blue
-D TFT_RGB_ORDER TFT_BGR=1 ; Colour order Blue-Green-Red

These problems occur because different suppliers wire the display pixels in different ways.