Bodmer/TFT_eSPI #2940: Sprites on 320x240 Display

SkydiveH, 2023-10-30:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. A description of the problem and the conditions that cause it to occur
2. IDE (e.g. Arduino or PlatformIO)
3. TFT_eSPI library version (try the latest, the problem may have been resolved!) from the Manage Libraries... menu
4. Board package version (e.g. 2.0.3) available from the Boards Manager... menu
5. Procesor, e.g RP2040, ESP32 S3 etc
6. TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.
7. Interface type (SPI or parallel)

Plus further information as appropriate to the problem:
1. TFT to processor connections used
2. A zip file containing your setup file (just drag and drop in message window - do not paste in long files!)
3. A zip file  containing a simple and complete example sketch that demonstrates the problem but needs no special hardware sensors or libraries.
4. Screen shot pictures showing the problem  (just drag and drop in message window)

The idea is to provide sufficient information so I can setup the exact same (or sufficiently similar) scenario to investigate and resolve the issue without having a tedious ping-pong of Q&A.

DO NOT paste code directly into the issue. To correctly format code put three ticks ( ` character on key next to "1" key) at the start and end of short pasted code segments to avoid format/markup anomolies. [See here:](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)

Example output:

```
  Serial.begin(115200);
  tft.init();
```

SkydiveH, 2023-10-30:
Hello sir,
I'm using the library very extensive and it is working perfectly for my Purpose. The only problem I have is if I want to create a sprite greater than 60x260. I tried 80x260 and when I request if he has created the sprite he denies it, he brings back 0 . I found that there is a limit for sprites greater than appr. 200x100.  60x260 works, 80x260 don't.
I'm using the Arduino IDE 2.2.1 with the latest update of the TFT_eSPI. Is there a possibility to create sprite bigger then the mentioned above or so. I don't not know where the effective limit is and how to change that. The biggest sprite I wanted to create was 240x320 and that don't worked, so I have to distribute it in separate sprite. That worked but is a not so satisfactory . 
The processor I'm using is a ESP32 Dev Kit 4 with 16M. Memory space should be not the problem but how to enable sprite greater than the 200x100?
Thank you 
Skydive

Bodmer, 2023-10-30:
To create larger sprites the board will need to have onboard PSRAM and that must be enabled in the IDE. The library will then use PSRAM for all sprites.