Bodmer/TFT_eSPI #2960: LilyGo_T_Display_S3 not working well with 2.5.33 ok with 2.5.0

beding, 2023-11-07:
Only raise issues for problems with the library and/or provided examples. Post questions, comments and useful tips etc in the "Discussions" section.

To minimise effort to resolve issues the following should be provided as a minimum:
1. fillscreen(TFT_BLACK) does not fill the complete scree.
2. drawstring() does not use given coordinates.

2. IDE Arduino 2.2.1
3. TFT_eSPI library version is OK, latest 2.5.33 has these problems.
4. Board package ESP32S3 Dev Module (i cannot see the version, but i have the latest (7 November 2023)
5. Procesor ESP32 S3
6. [TFT driver (e.g. ILI9341), a link to the vendors product web page is useful too.](https://www.lilygo.cc/products/t-display-s3)
7. Interface type (SPI)

Plus further information as appropriate to the problem:
[T-Display-S3-helloWorld.zip](https://github.com/Bodmer/TFT_eSPI/files/13278906/T-Display-S3-helloWorld.zip)

Regards,
Bert

Bodmer, 2023-11-07:
Which setup file are you using, paste content here if it is not an existing library file.

Have you been using the pre-configured library 2.5.0 customised by LilyGo here? :
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TFT_eSPI

When the library is updated from this repository have you configured it to use the correct setup file?

When using the latest library from here, run Read_User_Setup and check output is as expected. Post a copy here.

beding, 2023-11-07:
Yes, that preconfigured file works.

But if I update the library and select the display again, it no longer works
as it should.

 

So, preconfigured   V2.50

// #include <User_Setup.h>           // Default setup is root library folder

#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>

 

While typing this email, I do the same again.

Now updated the lib in Arduino ide to 2.5.33

And edited the same 2 lines.

Now testing the sketch again, . hmm, I see no problems.

 

So, I don't know what happened yesterday, I guess it must be that I have
configured the wrong display.

I ow you an apology.

 

And thanks for the quick response.

 

Regards,

Bert

 

 

 

 

 

 

 

 

Verzonden vanaf Outlook voor Android <https://aka.ms/AAb9ysg> 

  _____  

2.5.33 ok with 2.5.0 (Issue #2960) 

 

https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TFT_eSPI

view it on GitHub, or
KBLRAVCNFSM6AAAAAA7A65PQSVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTOOJZGYYTK
MZUG4> unsubscribe.
<https://github.com/notifications/beacon/AMAV2NPAJD6UROQ77NUDRBTYDKBLRA5CNFS
M6AAAAAA7A65PQSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

Bodmer, 2023-11-08:
OK, that is good. Glad to hear it is working.

closed 2023-11-08 by Bodmer