Bodmer/TFT_eSPI #2974: PushToSprite - SetSwapBytes - Transparency - Issue

michaimmendoerfer, 2023-11-14:
I have a fullscreen (240/240) sprite as a TFT-Buffer.
SetSwapBytes = true and everything is fine.

I have a 100x100 sprite (TFTGauge) also with SwapByte=true which i push to the primary buffer with pushToSprite with transparency set (which is 0x2945 in this case)...

TFTGauge.pushToSprite(&TFTBuffer, 22, 25, 0x2945);

What happens: 

case 1: 
TFTBuffer: SetSwapByte = true
TFTGauge: SetSwapBytes = true

--> correct colors, no transparency

case 2:
TFTBuffer: SetSwapByte = true
TFTGauge: SetSwapBytes = false

--> wrong colors, correct transparency

Maybe the transparency value must be swapped too?

Bodmer, 2023-11-15:
Provide a complete example so I can see exactly what the issue is.

michaimmendoerfer, 2023-11-15:
`
//Fullscreen-Buffer
TFTBuffer.createSprite(240,240);
TFTBuffer.setSwapBytes(true);
 
//small button to be pushed into Buffer with transparency color 0x2945 
TFTGauge1.createSprite(100,100);
TFTGauge1.setSwapBytes(false);

//Background-image
TFTBuffer.pushImage(0,0, 240, 240, JeepifyBackground);

//put button in it 4 times

TFTGauge1.pushToSprite(&TFTBuffer, 22, 25, 0x4529);      //working (swapped the transparency-bytes)
TFTGauge1.pushToSprite(&TFTBuffer,118, 25, 0x2945);     //not working
TFTGauge1.pushToSprite(&TFTBuffer, 22,115, 0x4529);
TFTGauge1.pushToSprite(&TFTBuffer,118,115, 0x2945);`

for me it is solved, but I had to manually change (byte-switch) the transparency-color...

Just read a bit... Is this such a Gulliver Endian-Thing?

closed 2023-11-15 by Bodmer