Bodmer/TFT_eSPI #2985: Colors issues with ILI9341_2 on ESP32-2432S028R with 2 USB ports

TheNitek, 2023-11-18:
I am using bodmer/TFT_eSPI@^2.5.33 on a ESP32-2432S028R with 2 USB ports, which seems to have another variant of the ILI9341_2 (compared to the 1 USB variant).
In general it works well (just needs TFT_INVERSION_ON)

Only issue is that the colors are way of, which looks like very poor gamma settings:

[image]

While I don't have a clue how gamma works on the ILI9341, I just started experimenting and noticed that this highly improves the picture quality:

```
  tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
  tft.writedata(2);
  delay(120);
  tft.writecommand(ILI9341_GAMMASET); //Gamma curve selected
  tft.writedata(1);
```

[image]

Does the gamma table maybe need a change because of the inversion?

harrytrinh9, 2023-11-25:
I have same issue with @TheNitek 
The image display actually very poor quality after call inverted display function.
[image]

zehn-cat, 2023-11-26:
I had the same problem with the same board. I get normal colors by commenting out these two lines from ILI9341_Init.h:
writecommand(0xE0); //Set Gamma
writecommand(0XE1); //Set Gamma

harrytrinh9, 2023-11-26:
Thanks @zehn-cat so much !
Your solution is work.
[image]

hello-world-dot-c, 2023-11-29:
@TheNitek This works great for me, thanks! How did you come up with this sequence? Did you find it somewhere?

Bodmer, 2023-12-05:
Unfortunately the TFT screens vary in supplier and this screen appears to need different settings. Also some screens appear to have cloned chips fitted that are not OEM.

So some tweaking may be needed in these cases.